bbcmicrobit/micropython #498: Until v1.0 is out of beta and release docs should point to v0.9

microbit-carlos, 2018-03-08:
I believe this was already done by @deshipu, but perhaps it has changed with the latest merge.

http://microbit-micropython.readthedocs.io/ should point to http://microbit-micropython.readthedocs.io/en/0.9/

At the moment is likely to be pointing to master, things like the Image docs contain `fill` and `blit`, which are not present in v0.9.

DavidWhaleMEF, 2018-09-09:
A more important solution here, is that each version of each specific editor, should link to the specific version of readthedocs, that are relevant for the version of MicroPython baked into the .hex files that it generates.

This has implications for the following projects:

mu as from codewith.mu
PythonEditor hosted at python.microbit.org

We probably already have tickets in those two projects related to this issue.

microbit-carlos, 2018-10-09:
We have now release the final 1.0.0 and that is also deployed on python.microbit.org, so the docs are in sync with the latest version.

We do have a tag for 0.9, and we are planning to make sure the versions of the python editor with older versions of MicroPython will point to the right versions of the docs. 

@deshipu @dpgeorge Looks like we don't have 1.0 tagged in the docs, could we create a "readthedocs tag" (not sure what's correct term here) to point to the GH tag for 1.0.0?
[image]

dpgeorge, 2018-10-12:
> Looks like we don't have 1.0 tagged in the docs, could we create a "readthedocs tag" (not sure what's correct term here) to point to the GH tag for 1.0.0?

I don't have access to administer RTD.  I don't know who created it, maybe @ntoll?

ntoll, 2018-10-12:
Hi Folks,

If you give me your usernames on ReadTheDocs I'll add you as maintainers and commit root-privilege suicide. The docs are your responsibility and I'll have nothing more to do with them.

@dpgeorge I'm guessing that since you're on GitHub "George 2.0" hasn't made an appearance yet? ;-)

microbit-carlos, 2018-10-16:
Thank you Nicholas, and everybody else like @markshannon, @dpgeorge, @deshipu (and many others!) for all the amazing work you have done in the docs.

I don't know @dpgeorge's ReadTheDocs username (other than assuming it might be the same as GitHub: https://readthedocs.org/profiles/dpgeorge/), but if you give my account root access I can then do the same for anybody else that might need it.
My username is: `microbit-carlos`

ntoll, 2018-10-29:
OK... added Carlos and Damien to RTD. Feel free to remove me.

ntoll, 2018-10-29:
Sorry it's taken so long to do this... the past two weeks have been, umm, interesting/hectic.

dpgeorge, 2018-10-30:
> OK... added Carlos and Damien to RTD. Feel free to remove me.

Thanks @ntoll, I've now removed you :)

microbit-carlos, 2018-10-30:
Thank you @ntoll.

With respect to the original issue, the readthedocs default landing version is the `latest` build, as that will contain the best documentation to match the released v1.0.0. While v1 was released over a month ago, there are several commits in the master branch since the v1.0.0 tag to bring the docs up-to-date (and no MicroPython changes yet), so showing `latest` by default is the best option right now.

We can revisit what version of the docs show by default once any MicroPython changes lands on master.

closed 2018-10-30 by microbit-carlos