bbcmicrobit/micropython #499: Using radio.send/receive in a tight loop results in "data packet received is not a string"

romilly, 2018-03-14:
Issue #357 is marked closed, but the problem is still alive and easily reproducible.

#On micro:bit A
import radio

radio.on()
while True:
    radio.send('flash')

#On micro:bit B
import radio
radio.on()
while True:
    try:
        incoming = radio.receive()
    except Exception as e:
        print('aargh %s' % e)
        radio.off()
        radio.on()
        continue
    if incoming == 'flash':
        print('blink')
--------------------------
You'll see lots of blinks, and an 'aargh data packet received is not a string' after every few blinks.

The radio.off(); radio.on() clears the error, allowing normal reception to resume.

ZanderBrown, 2018-03-14:
Yes I've seen this before, I believe this is partly a 'bandwidth' problem (too much stuff flying through the air)

DavidWhaleMEF, 2018-03-14:
I've seen this many times. My fix is to put a try/except block and turn the radio off and on again(!!)

I had a hunch that when a corrupted packet comes in, something in receive fails with a ValueError, but it leaves that payload in the receive queue, so trying to receive again after that always causes a failure. Thus the workaround of turning it off and on again to flush the queue.

One point to note is that I recall @finneyj telling me that the hardware/stack should check the checksum and reject payloads where the checksum does not match. So I'm surprised that corrupted payloads are even getting into the receive queue. @dpgeorge does the Python wrapper code do something different to circumvent this checksum checking?

I first saw this when coding 7 micro:bits to transmit in a tight loop. This causes a high rate of failures in the receiver. In order to get 196 micro:bits working for a big collaborative project, I timeslotted all communications from a master time source to *avoid* collisions, with the added safety net of turning the radio off and on again if radio.receive() threw a ValueError.

romilly, 2018-03-14:
I came across this when doing a test of 25 micro:bit fireflies
<http://microbit-micropython.readthedocs.io/en/latest/tutorials/radio.html#fireflies>.
I eventually found the radio off/on trick workaround*. *

I'll do a PR for the micro:bit MicroPython docs explaining the problem and
the workaround. I'll also do a blog post :)

dpgeorge, 2018-03-15:
> the hardware/stack should check the checksum and reject payloads where the checksum does not match. So I'm surprised that corrupted payloads are even getting into the receive queue. @dpgeorge does the Python wrapper code do something different to circumvent this checksum checking?

The code does check the CRC and does reject packets if they fail the CRC.  If something does get through which is not a string then it will raise a `ValueError` when calling `radio.receive()`, but the invalid packet is popped off the queue so it should be possible to continue to operate without doing a `radio.off()` follewed by `radio.on()`.  So I'm not sure what is going on here...

romilly, 2018-03-15:
@dpgeorge Is there a circular packet buffer? My guess was that I was seeing
a buffer over-run corrupting the header after correct reception.

dpgeorge, 2018-03-16:
I had a look into this issue and I think I found the problem: there was a race condition with the hardware writing a new received packet into RAM, while at the same time the software popping a previous packet off the RX queue and updating the pointer where the hardware should write the next packet (NRF_RADIO->PACKETPTR).  I fixed it in ee6ddc81c43d66efbb249d77822dfd9f286a104b.  See that commit's comments for further details.

dpgeorge, 2018-03-16:
It should now be possible to do the following without any issues:

On microbit A:
```python
import radio
radio.on()
while 1:
    radio.send('hello')
```
And on microbit B:
```python
import radio
radio.on()
while 1:
    data = radio.receive()
    if data:
        print(data)
```

closed 2018-03-16 by dpgeorge

DavidWhaleMEF, 2018-03-16:
Thanks @dpgeorge this particular bug has been in the wild for some time and we all just got in the habit of 'turning it off and on again'!!! Thanks for finding and fixing it. cc @jaustin @carlosperate

romilly, 2018-03-16:
Great work, thanks!