bbcmicrobit/micropython #505: Is there any way to call a library like any other development environment?

hnwangkg-ezio, 2018-03-21:
Is there any way to call a library like any other development environment?

from microbit import *
import GUI

I want to add a library of import GUI as above. I haven't worked hard for a long time and I didn't find a suitable method. The editor I use is Mu. If so, then my project will be very clean.

deshipu, 2018-03-21:
You can upload python modules to the micro:bit using Mu's "files" tab. Then you can import them.

MrYsLab, 2018-03-21:
The procedure is not exactly intuitive. Here is a guide on how to do this:
https://microbit-playground.co.uk/howto/add-python-module-microbit-micropython

I find it easier to just copy the code I wish to import to the top of my file if the import is fairly simple.

rhubarbdog, 2018-05-10:
you may find the command ufs (microfs) an easier to use command line option. install it on debian linux with  the following `sudo -H pip3 install microfs` . You then just type `ufs put GUI.py`