bbcmicrobit/micropython #506: mciropython need to support more device such as dht11

yonghuming, 2018-03-22:
mciropython need to support more device

rhubarbdog, 2018-05-18:
Hi
I've just ported a DHT11 sensor from raspberry pi.  it would appear that the micro:bit runs too slow for 1 wire protocol devices like the DHT11. i use an AM3220 over the i2c bus for temperature and humidity sensing you can download the relevant code from https://github.com/mcauser/microbit-am2320

rhubarbdog, 2018-11-29:
I've done some work on this sensor class, it's still not working, but it's close, it receiving 35 out of 40 bits. I've written the time critical code in `arm Thumb Assembly` the .read method is too slow at switching between triggering the sensor and starting to grab the bit stream received from the sensor.
The repository is https://github.com/rhubarbdog/microbit-dht11 please feel free to try and get this working i'm going to research speeding up my microbit/micropython

rhubarbdog, 2018-11-29:
have you got examples of further sensors you would like to see running on microbit?

rhubarbdog, 2018-12-01:
hi
the sensor is now working you can download it from https://github.com/rhubarbdog/microbit-dht11