bbcmicrobit/micropython #509: micro:bit freezes when used with PWM (write_analog)

eudaemon, 2018-03-31:
Hi,

I have been using the latest http://python.microbit.org/v/1 and experienced micro:bit (receiver side) freezing with PWM (write_analog). Could someone take a look at this code and find out what is wrong?

* Controller Code
```python
from microbit import *
import radio

# The unique channel number [0, 100]
my_channel = 19
# This is the usual magnitude max value of x and y
mmax = 1072
# This is theoretical max value of microbit max
amax = 1023
# Wait time in ms
wait_time = 100
# X atten. It is hard to drive. So atten is better.
xatten = 2.0

radio.on()
radio.config(channel=my_channel)    # Choose your own channel number
radio.config(power=7)               # Turn the signal up to full strength

display.scroll("S"+str(my_channel), delay=100, wait=False, loop=True)

# Event loop.
while True:
    x = min(mmax, max(-mmax, accelerometer.get_x() / xatten))
    y = min(mmax, max(-mmax, accelerometer.get_y()))
    # This has to be this way to be intuitive turning. Swap(Lw,Rw)
    Lw = (x + mmax) / (mmax + mmax)
    Rw = 1 - Lw
    # It has to be negative y to be more intuitive. y => -y
    y2 = -y / mmax * amax
    Lw = min(amax, max(-amax, int(Lw * y2)))
    Rw = min(amax, max(-amax, int(Rw * y2)))
    
    radio.send(str(Lw) + ',' + str(Rw))
    sleep(wait_time)
```

* Receiver Code
This is the problematic code where it runs for a while and then freezes randomly.
```python
from microbit import *
import radio

# The unique channel number [0, 100]
my_channel = 19
# The smoothing constant. Higher=Low_Reaction(1.0), Lower=Smoother_control(0.1)
smooth = 0.8
# Wait time in ms. Should be smaller than the sender's
wait_time = 100 / 2

radio.on()
radio.config(channel=my_channel)    # Choose your own channel number
radio.config(power=7)               # Turn the signal up to full strength

display.scroll("R"+str(my_channel), delay=100, wait=False, loop=True)

# Event loop.
Lw = 0
Rw = 0
while True:
    try:
        incomingRaw = radio.receive()
        if incomingRaw is None:
            sleep(wait_time / 4)
            continue
        (LwR, RwR) = incomingRaw.split(',')
        print('L,R:' + LwR + ',' + RwR)
        Lw = int(int(LwR) * smooth + Lw * (1 - smooth))
        Rw = int(int(RwR) * smooth + Rw * (1 - smooth))
        if (Lw < 0):
            pin0.write_digital(0)
            pin8.write_analog(-Lw)
        else:
            pin8.write_digital(0)
            pin0.write_analog(Lw)
        if (Rw < 0):
            pin1.write_digital(0)
            pin12.write_analog(-Rw)
        else:
            pin12.write_digital(0)
            pin1.write_analog(Rw)
    except Exception as e:
        print("Exception %s" % str(e))
    sleep(wait_time)
```

Thanks,

carlosperate, 2018-04-02:
Hi @eudaemon,
Thanks for the report, there was some PWM and radio issues that have been fixed but not officially released on the online editor.
Could you try the code in this beta version of the editor containing the latest version of MicroPython? http://python.microbit.org/v/beta

eudaemon, 2018-04-03:
It solves the problem! Thanks.

closed 2018-04-03 by eudaemon

microbit-carlos, 2018-04-03:
Glad it worked, thanks for confirm it.