bbcmicrobit/micropython #511: radio.receive_full() not present in radio module

descartez, 2018-04-16:
I've tried to use radio.receive_full() as found here[http://microbit-micropython.readthedocs.io/en/latest/radio.html#radio.receive_full].

And I keep getting the following error:
`AttributeError: 'module' object has no attribute 'recieve_full'`

Has there been a change to this that I am unaware of? As far as I can tell, no one else is having this issue. I am currently using Mu, though I still have the same problem when I use the web coding environment as well.

ukBaz, 2018-04-16:
I think the issue is that you are ahead of the game! The `receive_full` has only just been implemented #410 and has not made it into tools like Mu.

Mu does allow you to update the hex file if you want to use a newer version:
https://github.com/mu-editor/mu/issues/435

dpgeorge, 2018-04-16:
Mu was updated to the latest MicroPython firmware on 19th March 2018; see the commit https://github.com/mu-editor/mu/commit/e66585f65895b9f591f8a855b51518eb12752098.  This version includes the `receive_full()` function.  Maybe you are using an old version of Mu?

dpgeorge, 2018-04-17:
> beta15 was Feb 21st

I see.  I'm not sure how the release cycles work with Mu but I guess it'll be updated soon.

rhubarbdog, 2018-05-18:
I use the command uflash. it allows you to specify a more up to date firmware to use as a base .hex file.
installing uflash
`sudo -H pip3 install uflash`
downloading a more up to date firmware
https://forum.micropython.org/viewtopic.php?f=17&t=4161&p=25151&hilit=firmware#p25151

microbit-carlos, 2018-08-20:
radio.receive_full() was implemented in September 2017 (https://github.com/bbcmicrobit/micropython/commit/e6f9ab13baa86cab58cbd093c7f2adf91ce56af7), so it would have been present on any version of Mu from 1.0.0.beta.12: https://github.com/mu-editor/mu/commit/841f9a996ab479bb90a8887b1ab0ea09ab050cf8

However since there wasn't official installers available until beta 16 (Feb/March 2018), then the only way to try the beta version at the time this issue was opened was by running from master branch. Taking that in consideration, it iss very likely most users were using and older version of Mu at that point.

Since Mu v1.0.0 has been released for all platforms now this function should be available to most Mu users.

closed 2018-08-20 by microbit-carlos