bbcmicrobit/micropython #512: microbit driver

ouki-wang, 2018-04-20:
When I uninstalled microbit driver and try re-installing, I was surprised that the PC cannot recognise the devloping board. Now, although I have installed the driver, but the developing cannot be used and the serial port number cannot be recognised either. I have tried uninstall and re-install the driver in different computers for many times, but still failed. Please help me to solve this problem.

dpgeorge, 2018-04-20:
Which driver are you talking about?  In most cases (Mac, Windows 10, Linux) you don't need a serial driver, or anything else.

carlosperate, 2018-04-20:
On Windows 10 the serial driver should be install automatically. On earlier version of Windows you might need to install that driver manually. More info can be found here: https://support.microbit.org/support/solutions/articles/19000022103-how-do-i-use-the-serial-port-with-a-micro-bit-on-windows

If you are having problems with the board you can also open a new support ticket on the same page and the support team can help you figure out the problem.