bbcmicrobit/micropython #514: fails to link with arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 12/1

shenki, 2018-05-01:
I see this on my Ubuntu system.

```
$ /usr/bin/arm-none-eabi-g++ --version
arm-none-eabi-g++ (15:6.3.1+svn253039-1build1) 6.3.1 20170620
```

```
GCC version is: 6.3.1
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /home/joel/dev/micropython-microbit/build/bbc-microbit-classic-gcc-nosd
[5/5] Linking CXX executable source/microbit-micropython
FAILED: source/microbit-micropython 
: && /usr/bin/arm-none-eabi-g++  -fno-exceptions -fno-unwind-tables -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment -Wl,-wrap,main -mcpu=cortex-m0 -mthumb -T"/home/joel/dev/micropython-microbit/yotta_targets/bbc-microbit-classic-gcc-nosd/CMake/../ld/NRF51822.ld" -Wl,-Map,source/microbit-micropython.map -Wl,--start-group source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/gc.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modaudio.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modantigravity.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modmusictunes.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/parsenum.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/microbituart.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/microbitspi.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/sequence.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modspeech.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/extmod/utime_mphal.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/frozenmod.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/moduerrno.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objint_longlong.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/asmthumb.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/modstruct.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/builtinevex.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objreversed.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objslice.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/persistentcode.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/obj.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/nlrx64.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/emitinlinextensa.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objfilter.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/asmxtensa.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/modcollections.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/lib/pwm.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objsingleton.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objrange.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/nlrthumb.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/builtinimport.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/showbc.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objenumerate.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/scheduler.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/microbitpin.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objclosure.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/modsys.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/events.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/builtinhelp.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/compile2.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/argcheck.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objstringio.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modthis.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/malloc.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modmusic.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objtuple.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/vm.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/formatfloat.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/nlrsetjmp.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/lib/sam/sam.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/modcmath.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/emitbc.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objobject.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/lib/neopixel.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modlove.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objint_mpz.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/main.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/mphalport.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modneopixel.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/asmx64.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/modio.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objlist.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objmodule.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/warning.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/compile.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modmicrobit.cpp.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/emitglue.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/parse.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objcell.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/fileobj.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/microbit/modos.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/nativeglue.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/py/objexcept.c.o source/CMakeFiles/microbit-micropython.dir/home/joel/dev/micropython-microbit/source/lib/mp-readline/readline.c.o source/CMakeFiles/microbit-micropy [... cut ...]
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v6-m/libstdc++_nano.a(new_op.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/thumb/v6-m/libstdc++_nano.a(new_handler.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

steven-eckhoff, 2018-06-08:
Did you every fix this? I have the same.

dpgeorge, 2018-06-09:
Try removing completely the `build/`, `yotta_modules/` and `yotta_targets/` directories and then rebuilding.

shenki, 2018-06-12:
This warning (I trimmed the paths to make it easier to read) is due to the way the Ubuntu cross compiler is configured:
```
arm-none-eabi/bin/ld: error: arm-none-eabi/lib/crt0.o: Conflicting CPU architectures 12/1
arm-none-eabi/bin/ld: failed to merge target specific data of file arm-none-eabi/lib/crt0.o
```

```
$ objdump -f /usr/lib/arm-none-eabi/newlib/crt0.o | sed -n '3p'
architecture: armv4, flags 0x00000011:
```
The crt0 it's trying to link in is built for ARM(v4), but a m0 machine only supports thumb.

When building for `-mcpu=cortex-m0`, the compiler looks in `thumb/v6-m` for libraries:
```
$ arm-none-eabi-gcc -print-multi-directory -mcpu=cortex-m0 
thumb/v6-m
```
However, we don't have a `crt0.o` in `/usr/lib/arm-none-eabi/newlib/thumb/v6-m`. There is one in `thumb/`, however gcc doesn't seem to fall back to that, and instead falls back to /usr/lib/arm-none-eabi/newlib/crt0.o.

I did some experiments to work out what was going on, before realising that Ubuntu appears to have fixed the issue since I reported it. I'm running libnewlib-arm-none-eabi 2.4.0.20160527-4. Ubuntu bionic has 2.4.0.20160527-3, which I think was the broken version.

You could try to install the version of libnewlib-arm-* from Ubuntu cosmic?

rojer, 2018-07-09:
this could be a Ubuntu 18.04 bug https://bugs.launchpad.net/ubuntu/+source/gcc-arm-none-eabi/+bug/1767223
please pile on :)

temporaryaccount, 2018-07-13:
> You could try to install the version of libnewlib-arm-* from Ubuntu cosmic?

Yes, I can confirm fixing the issue by installing the newer packages.

Just get the  [libnewlib-dev (2.4.0.20160527-4)](https://packages.ubuntu.com/cosmic/all/libnewlib-dev/download) and [libnewlib-arm-none-eabi (2.4.0.20160527-4)](https://packages.ubuntu.com/cosmic/all/libnewlib-arm-none-eabi/download) *.deb files, then install them:

`sudo dpkg -i dpkg -i libnewlib-dev_2.4.0.20160527-4_all.deb libnewlib-arm-none-eabi_2.4.0.20160527-4_all.deb`

johnthagen, 2018-08-24:
As of today, @temporaryaccount's instructions worked for Ubuntu 18.04.1 (but the versions provided at the links have been updated).

```
$ sudo dpkg -i libnewlib-arm-none-eabi_3.0.0.20180802-2_all.deb libnewlib-dev_3.0.0.20180802-2_all.deb
```