bbcmicrobit/micropython #515: The links on this page don't make sense

rhubarbdog, 2018-05-10:
This page is all about using REPL at the bottom there are operating system specific links these all go to the firmware flashing operating system page. You need to setup a new page for REPL operating system specifics. if it were there as a blank with titles i could help you with linux information

dpgeorge, 2018-05-11:
Yes you are right that section at the end of that page doesn't make sense.  In fact that whole page doesn't flow very well.  If you have suggestions how to fix it please post them!

rhubarbdog, 2018-07-28:
I've read that page and REPL is quite advanced and there isn't that much that can be said except adding some text regarding using REPL in the development cycle.  It wasn't until i bought a pyboard with no screen that I really appreciated what REPL gives.  There should perhaps be a mention of debugging and the support it gives for the print statement.
With regards to linux and the screen program you need to specify the baud rate of 115200 so you would type `screen /dev/ttyACM0 115200` on debian and raspberry pi the first argument is `/dev/ttyACM0`.  Sometimes screen just immediately terminates, that's because the file /dev/ttyACM0 is in use by another process, either wait for five minutes or try the following `sudo lsof /dev/ttyACM0` followed by killing the offending process by typing `sudo kill -term <process-id>` in both cases you have to use sudo as the offending process may be owned by root. Notes on exiting screen I used to type control-D (EOF), that would simply hang screen allowing me to re-attatch at a later date typing `screen -r` which isn't what i wanted in actuality I wanted to terminate screen for which you type control-A \ y

rhubarbdog, 2018-07-29:
it would appear on a mac you type the following to get REPL working `screen /dev/tty.usbmodem1422 115200`

microbit-rosslowe, 2018-10-04:
@rhubarbdog does this proposal make sense? https://github.com/bbcmicrobit/micropython/pull/570/commits/af60e875e24024b1ad96cd2fd6d937f36e1c5785

microbit-carlos, 2018-10-19:
PR #570 has been merged. Thank you @rhubarbdog for the report and @microbit-rosslowe thank you for the PR!
I'll close this issue, but if more info or clarifications are needed please feel free to reopen or create a new issue.

closed 2018-10-19 by microbit-carlos