bbcmicrobit/micropython #516: Method to override the value error if a display pin is used for I/O when the display is on

tim1mw, 2018-05-19:
I have some code that requires one more analogue I/O pin that is available when the display is turned on. I just wrote the code to use a pin which doubles up with the display regardless. With earlier versions of interpreter, this code would work just fine. There is some display corruption, but not so much that the output I need to show is unreadable. However, the latest MicroPython hex now gives an error when I try to run my code:

ValueError: Pin X in display mode

I appreciate that most people would want the sanity check here, but it would be really handy if there were some kind of override so that code such as mine can continue to work. I really want to do this! The code I'm using has been running reliably for about 8 months now with no obvious adverse effects on the MicroBit. I can work around this for now by using an older version of the MicroPython hex, so it's not an urgent problem, but there may be other use cases where this is desirable.

I've noticed this primarily because I upgraded from Mu 0.9.13 to the 1.0.0 beta a while back, which comes with a more recent version of the hex file. I'm assuming this commit:

https://github.com/bbcmicrobit/micropython/pull/345/commits/ce36e0dea1c23f0608493978c839707386f34ce6

Is the point where the sanity check was added.

I suppose another way to deal with this would be by allowing the display to be partially enabled, so that pins which are shared with the display can be released while retaining some display function, but I can no idea if the Micro:Bit hardware can support that.

dpgeorge, 2018-05-22:
While your case does seem to be a bit of an outlier I agree it would be good to be able to override the pin check.  Do you have an idea of how the API would look, how the user would select to override it?

rhubarbdog, 2018-08-20:
have you thought of saving the current screen (if needed) using `display.get_pixel()`
switch the display off using `display.off()` 
read/write data to the pin
switch the display back on `display.on()`
restore the screen using `display.set_pixel()`

tim1mw, 2018-08-20:
I suspect that would make the screen flicker, since I need to poll all the input pins repeatedly at a fairly high rate. However, I might give it a go just to see what the effect is.

Re an API to make this work (sorry I missed that comment), my thought was that there would be a simple flag or single method call that could be used at the start of the code to disable the sanity check.

deshipu, 2018-08-20:
Have you considered other options, such as using a pin expander or an external ADC chip? You know, ones that are not hacks.

tim1mw, 2018-08-20:
I really don't want to complicate the hardware, to me turning off the sanity check is far simpler and it seems like a useful, if unusual feature. If the answer is "no we're not doing this", then I'll just stick with the old microcode, the project I have works perfectly and I'm reluctant to fiddle with it.