bbcmicrobit/micropython #517: Use Neopixels on more pins?

sofiania, 2018-05-22:
Can 3 neopixels with 8 LEDS be connected simultaneously to say pin 0, pin 1, pin 2?

dpgeorge, 2018-05-22:
Yes that should be possible.  Give it a try!

rhubarbdog, 2018-07-28:
be careful, 3*8 neopixel may pull 2 amps of power, you'll need an external power supply for your LED's

rhubarbdog, 2018-08-03:
Try any of these pins they are all PWM capable and aren't connected to anything else
pin0, pin1, pin8, pin2, (pin13, pin14, pin15) & pin16
You can use any of the pins which support write_analog, but you would need to start disabling display, not using the buttons etc

microbit-carlos, 2018-08-20:
@sofiania I assume you've been able to run neopixels on several pins without issues, so I'll close this one.
However, please let us know if you've encountered any problems and we can reopen it.

closed 2018-08-20 by microbit-carlos