bbcmicrobit/micropython #523: Unable to build on Raspberry Pi 2B

Rick1997, 2018-06-06:
I am trying to generate python code on a Raspberry Pi and upload it to the Microbit. I get the following error when trying to build on a Raspberry Pi.

```
pi@raspberrypi:~/micropython $ yt build
info: generate for target: bbc-microbit-classic-gcc-nosd 0.2.3 at /home/pi/micropython/yotta_targets/bbc-microbit-classic-gcc-nosd
CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/pi/micropython/yotta_targets/mbed-gcc/CMake/toolchain.cmake:78 (cmake_force_c_compiler)
  toolchain.cmake:8 (include)
  CMakeFiles/3.7.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:78 (project)

CMake Deprecation Warning at /usr/share/cmake-3.7/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/pi/micropython/yotta_targets/mbed-gcc/CMake/toolchain.cmake:79 (cmake_force_cxx_compiler)
  toolchain.cmake:8 (include)
  CMakeFiles/3.7.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:78 (project)

GCC version is: 5.4.1
suppressing warnings from ble-nrf51822
suppressing warnings from nrf51-sdk
suppressing ALL warnings from mbed-classic, ble, ble-nrf51822 & nrf51-sdk
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/micropython/build/bbc-microbit-classic-gcc-nosd
[6/147] Building CXX object source/CMakeFiles/microbit-m...ome/pi/micropython/source/microbit/microbitcompass.cpp.o
FAILED: source/CMakeFiles/microbit-micropython.dir/home/pi/micropython/source/microbit/microbitcompass.cpp.o
/usr/bin/arm-none-eabi-g++  -DYOTTA_MODULE_NAME=microbit-micropython -Dmicrobit_micropython_EXPORTS -Igenerated/include -I/home/pi/micropython -I/home/pi/micropython/yotta_modules/microbit-dal -I/home/pi/micropython/yotta_modules/mbed-classic -I/home/pi/micropython/yotta_modules/ble -I/home/pi/micropython/yotta_modules/ble-nrf51822 -I/home/pi/micropython/yotta_modules/nrf51-sdk -I/home/pi/micropython/inc -I/home/pi/micropython/inc/microbit -I/home/pi/micropython/yotta_modules/microbit-dal/inc/core -I/home/pi/micropython/yotta_modules/microbit-dal/inc/types -I/home/pi/micropython/yotta_modules/microbit-dal/inc/drivers -I/home/pi/micropython/yotta_modules/microbit-dal/inc/bluetooth -I/home/pi/micropython/yotta_modules/microbit-dal/inc/platform -I/home/pi/micropython/yotta_modules/mbed-classic/api -I/home/pi/micropython/yotta_modules/mbed-classic/hal -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal -I/home/pi/micropython/yotta_modules/mbed-classic/targets/cmsis -I/home/pi/micropython/yotta_modules/ble-nrf51822/source/btle -I/home/pi/micropython/yotta_modules/ble-nrf51822/source/btle/custom -I/home/pi/micropython/yotta_modules/ble-nrf51822/source/common -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/ble_radio_notification -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/ble_services/ble_dfu -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/common -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/device_manager -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/device_manager/config -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/ble/peer_manager -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/device -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/ble_flash -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/delay -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/hal -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/pstorage -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/drivers_nrf/pstorage/config -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/bootloader_dfu/hci_transport -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/crc16 -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/hci -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/scheduler -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/timer -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/util -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/fds -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/fstorage -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/libraries/experimental_section_vars -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/softdevice/common/softdevice_handler -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/softdevice/s130/headers -I/home/pi/micropython/yotta_modules/nrf51-sdk/source/nordic_sdk/components/toolchain -I/home/pi/micropython/yotta_modules/mbed-classic/targets -I/home/pi/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC -I/home/pi/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM -I/home/pi/micropython/yotta_modules/mbed-classic/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NRF51_16K_S110 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/crc16 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/scheduler -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/nordic_sdk/components/libraries/util -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s110_nrf51822_8_0_0 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/Lib/s130_nrf51822_1_0_0 -I/home/pi/micropython/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51_MICROBIT -I/home/pi/micropython/source -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -Wall -Wextra -fno-rtti -fno-threadsafe-statics -mcpu=cortex-m0 -mthumb -D__thumb2__ -Os -g -gdwarf-3 -DNDEBUG   -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_OPERATORS -DNRF51 -DTARGET_NORDIC -DTARGET_M0 -D__MBED__=1 -DMCU_NORDIC_16K -DTARGET_NRF51_MICROBIT -DTARGET_MCU_NORDIC_16K -DTARGET_MCU_NRF51_16K_S110  -DTARGET_NRF_LFCLK_RC -DTARGET_MCU_NORDIC_16K -D__CORTEX_M0 -DARM_MATH_CM0 -DNO_BLE -include "/home/pi/micropython/build/bbc-microbit-classic-gcc-nosd/yotta_config.h" -MMD -MT source/CMakeFiles/microbit-micropython.dir/home/pi/micropython/source/microbit/microbitcompass.cpp.o -MF source/CMakeFiles/microbit-micropython.dir/home/pi/micropython/source/microbit/microbitcompass.cpp.o.d -o source/CMakeFiles/microbit-micropython.dir/home/pi/micropython/source/microbit/microbitcompass.cpp.o -c /home/pi/micropython/source/microbit/microbitcompass.cpp
/home/pi/micropython/source/microbit/microbitcompass.cpp: In function 'void microbit_compass_init()':
[... cut ...]
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.4 (stretch)
Release:        9.4
Codename:       stretch
pi@raspberrypi:~/micropython $ cmake --version
cmake version 3.7.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
pi@raspberrypi:~/micropython $ ninja --version
1.7.2
pi@raspberrypi:~/micropython $ yotta --version
0.18.5
```

Is this because of my arm-none-eabi-gcc version? According to `apt-cache madison gcc-arm-none-eabi` this is the only version.

ZanderBrown, 2018-06-06:
Do you actually need to compile micropython or will uflash get the job done?

Rick1997, 2018-06-06:
>Do you actually need to compile micropython or will uflash get the job done?

You're right. I got it working with uflash. Thanks!

microbit-carlos, 2018-08-20:
As it looks like this is not an issue to look into anymore I'll close this one, otherwsie please feel free to reopen.

closed 2018-08-20 by microbit-carlos