bbcmicrobit/micropython #524: Can microbit's micropython run mpy files?

ouki-wang, 2018-06-28:


dpgeorge, 2018-06-28:
No, not at this point.

RobertLucian, 2018-09-06:
I know you @dpgeorge  already answered to @ouki-wang 's question, but what about now, after 2 years have passed? 

I precompiled a test python script using the version of micropython specified in here (I used the specified git hash) and then when I try to import it within the microbit I get an error telling me the module doesn't exist. Is it possible that I'm doing something wrong? 

Thanks!

dpgeorge, 2018-09-07:
For hints on getting it to load mpy files, see this comment: https://github.com/bbcmicrobit/micropython/issues/555#issuecomment-419351214

Hopefully this feature will make it into the main code soon.

pennyarcade, 2021-06-22:
Any news on this after another 2 years? ;-)