bbcmicrobit/micropython #525: Priority order to execute user code (embedded vs main.py)

microbit-carlos, 2018-07-10:
With the introduction of the filesystem we had the option to run a main.py file on startup. The current implementation prioritises a script saved at the known flash location (0x3E_0000) and I think we should probably reverse this order to prioritise main.py.

The reason this has come to light is because editors like Mu have started to flash the micro:bit using the filesystem (thanks @ntoll!), and we can have a situation where a micro:bit has MicroPython + embedded script, then a main.py file is then transferred, and the user would still see the old code running on the micro:bit.

It would also be really hard for an editor to determine if the micro:bit contains an embedded script in flash.

I've created a proposed PR for this change and gave it a quick test (embed script, load main.py and erase main.py), and seems to do the trick.

@dpgeorge what do you think? Would there be any other implications if this was changed?

dpgeorge, 2018-07-11:
> and we can have a situation where a micro:bit has MicroPython + embedded script,

How does this situation arise?  I thought Mu had a bare image in it (ie no appended script).  Does it come about when users download firmware with an appended script (maybe from the browser), and then reuse that device with Mu in filesystem-mode?

> Would there be any other implications if this was changed?

It does change behaviour, and likely there's someone out there relying on this behaviour (maybe unintentionally).  But I doubt it's a common thing, so it would be good to make it work more "as expected".  Since adding an appended script is only something that can be done when you flash the entire image, but changing main.py on the filesystem can (must!) be done after flashing the image, and can be done many times, it does make sense that main.py overrides the appended script.  After all, downloading main.py is an explicit, user-requested operation so they really must want to run the new main.py rather than the old one.

> It would also be really hard for an editor to determine if the micro:bit contains an embedded script in flash.

Using recent firmware it is actually very easy:
```python
import machine
have_appended_script = machine.mem16[0x3e000] == 0x504d
```
So this could be used as a way to warn the user that there exists an appended script.

But in the interest of simplicity it's probably a good idea to just make the filesystem main.py take precedence over any appended script.

ntoll, 2018-07-11:
That's a nice trick with the `machine` module.

microbit-carlos, 2018-07-11:
> How does this situation arise? I thought Mu had a bare image in it (ie no appended script). Does it come about when users download firmware with an appended script (maybe from the browser), and then reuse that device with Mu in filesystem-mode?

Yes, that would be one of the main ways this could happen. 

You can also force Mu to create a single hex using the embedded method and save it into your computer by flashing without a micro:bit attached.

And reading the flash address directly is a great trick!