bbcmicrobit/micropython #529: pin.is_touched() doesn't work :-(

ntoll, 2018-07-13:
The following is taken from a report by a Mu user. I've just given it a go and can confirm doesn't appear to work. I guess there's a bug which needs fixing before we can do a 1.0 :-(

------------------------------------------------------------------

Hi, when I try and flash the following code with Mu it displays the sad face but will not change to the happy face.

```
from microbit import *

while True:
    if pin0.is_touched():
        display.show(Image.HAPPY)
    else:
        display.show(Image.SAD)
```

I also tried with just uflash and it didn't work then either. However, it works fine when the .hex is downloaded off of the web client. Am I doing something wrong? I'm running Windows 8.1.

ntoll, 2018-07-13:
The reporter was using Mu beta.17 (which uses the last release of MicroPython for the micro:bit).

dpgeorge, 2018-07-14:
Thanks for the report, I can confirm the problem.  It looks like it's been broken for a while... it looks like the problem was introduced in 4c6160dccd7ab1f115d4963ebedf713c4f2bd311 over a year and a half ago!  The point is that the internal pull on the touch pin was changed from NOPULL to PULLUP in that commit (seems to be a mistake).  A previous commit 13bcee29144ae3732a87c66fa824b0cdaf9785fc put it to NOPULL.

Probably it wasn't reported because the touching still does work, but it's less sensitive with the extra PULLUP enabled.

In 2ecae7cc78fbbe5eba9cf16a5c84594d79343ed1 I fixed it to make it NOPULL, again.  It should now be more sensitive, but still remember that it won't work with really dry hands.

dpgeorge, 2018-07-14:
But using touch I now notice another problem with it: once a pin is in touch mode there doesn't look to be any way to get it out of touch mode.  So if you do `pin0.is_touched()` you can not do `pin0.read_digital()` or anything else until a hard reset.

dpgeorge, 2018-07-15:
> But using touch I now notice another problem with it: once a pin is in touch mode there doesn't look to be any way to get it out of touch mode.

This is addressed by PR #533.

dpgeorge, 2018-07-16:
> But using touch I now notice another problem with it: once a pin is in touch mode there doesn't look to be any way to get it out of touch mode.

Fixed by eeb9a2d2e4ec7d49354913fcafc990822430c662

closed 2018-07-16 by dpgeorge

ntoll, 2018-07-16:
This still doesn't work for me... :-( Using latest RC. cc/@microbit-carlos

ntoll, 2018-07-16:
OK... scratch that... un-plugging and plugging back in again made it work. No idea why... but that's good enough for me. ;-)