bbcmicrobit/micropython #535: Memory allocation error in Script that run in older versions of MicroPython

microbit-carlos, 2018-07-18:
From: https://github.com/mu-editor/mu/issues/565

> Mu 1.0.0.rc.1
> Windows 10
> 
> The attached code generates the following error "Line 16 Memory Error memory allocation failed allocating 2816 bytes". No error is generated when flashing the same code using Mu 0.9.13. The program is test code for a dual 4x7-seg display with MAX7219 driver.
> 
> As the program is quite long by micro:bit standards, it is possible that the new version of Mu just takes up slightly more memory than the old version and runs out of space.
> 
> [7-seg test no library with functions 2.zip](https://github.com/mu-editor/mu/files/2205154/7-seg.test.no.library.with.functions.2.zip)

Thank you @SteveJSmythe for the testing and reporting the issue!

microbit-carlos, 2018-07-18:
I have tested the code in 3 versions of MicroPython, nothing attached to the micro:bit:
* MicroPython v1.0.0-rc.1
    * From Mu v1.0.0.rc.1
    * I can replicate the allocation error:
        ```
        Traceback (most recent call last):
          File "__main__", line 16, in <module>
        MemoryError: memory allocation failed, allocating 2816 bytes
        ```
* MicroPython v0.9
    * From http://python.microbit.org/v/1
    * I can replicate a similar allocation error:
        ```
        Traceback (most recent call last):
          File "__main__", line 16, in <module>
        MemoryError: memory allocation failed, allocating 3840 bytes
        ```
* MicroPython from unknown commit (slightly newer than v0.9) 
    * From Mu v0.9.13
    * MicroPython from ntoll/uflash@fe26fe8
    * Not sure which MicroPython commit was used, based on the date (23rd of October 2016). It's likely slightly ahead of the MicroPython v0.9 (effc07c2571bf8c388bf25ee1392cc8a1f535aac 3rd October 2016), possibly 525d88c73f2b81d319efa3889cf5424ebfdc1770?
[image]
    * This version, as indicated in the bug report, doesn't get an allocation error.

dpgeorge, 2018-07-18:
I can confirm that with Mu v0.9.13 the 7-seg script works, but with v1.0.0-rc.1 it doesn't.

Unfortunately I don't think it can be fixed.  v1.0.0-rc.1 has about 350 bytes more heap than v0.9.13, so the issue would very likely be memory fragmentation.  In v0.9.13 I guess there just happened to be a large enough memory block free due to the pattern of allocations, and in v1.0.0-rc.1 the pattern of allocations has changed and there is no longer a region available.

But I think there's a bug in the 7 segment code that leads to this MemoryError: line 16 is allocating very large bytearray's, each time around the loop it allocates an array (256 * i) big, where i goes in the range 0 to 15 inclusive.  I think this is a mistake and it should be allocating just two bytes with the given value (256 * i).  Fixing that would definitely make the script work on v1.0.0-rc.1.

SteveJSmythe, 2018-07-18:
Thanks. I'll take a look at that loop.

dpgeorge, 2018-07-19:
Fixed by #539

closed 2018-07-19 by dpgeorge