bbcmicrobit/micropython #537: can't build on ubuntu 16

kylinpoet, 2018-07-18:
/home/ohwgy/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:229:31: error: in C++98 'centre' must be initialized by constructor, not by '{...}'
     Sample3D centre = { 0,0,0 };
                               ^
**### /home/ohwgy/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:230:29: error: in C++98 'best' must be initialized by constructor, not by '{...}'
     Sample3D best = { 0,0,0 };**
                             ^
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed

dpgeorge, 2018-07-18:
We can (and should) get rid of this include because it's no longer needed (since 6c36fa82b1ffb26f56ff17f64cb58eed0ff0401e).  But it could wait until after v1.0.0

kylinpoet, 2018-07-19:
Thanks for your replies.
I had check out the git source..
But When I use
**### make** .
it shows another error:
### **/home/ohwgy/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp: In static member function 'static Sample3D MicroBitCompassCalibrator::approximateCentre(Sample3D*, int)':
/home/ohwgy/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:229:31: error: in C++98 'centre' must be initialized by constructor, not by '{...}'
     Sample3D centre = { 0,0,0 };
                               ^
/home/ohwgy/micropython/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:230:29: error: in C++98 'best' must be initialized by constructor, not by '{...}'
     Sample3D best = { 0,0,0 };**

Is it something about the cmake???
when I use g++ -std=c++0x for g++ alias。。。
and Comment the "_newlib_version.h
But no helps..

dpgeorge, 2018-07-19:
The issue in MicroBitCompassCalibrator.cpp should probably be reported at its upstream source repository: https://github.com/lancaster-university/microbit-dal

But I don't see this problem building with arm-none-eabi-g++ 8.1.0

carlosperate, 2018-07-19:
> We can (and should) get rid of this include because it's no longer needed (since 6c36fa8). But it could wait until after v1.0.0

Yep, sound good.

@kylinpoet I cannot replicate either in Ubuntu 16.04.4 with arm-none-eabi-g++ 7.2.1.

How did you install your toolchain?
This are the scripts I've used to install them in mine:
https://github.com/carlosperate/microbit-dev-env/blob/master/install_gcc-arm.sh
https://github.com/carlosperate/microbit-dev-env/blob/master/install_cpp_toolchain.sh
https://github.com/carlosperate/microbit-dev-env/blob/master/install_yotta.sh

kylinpoet, 2018-07-20:
@carlosperate
Sorry for my poor English... I used the toolchain almost the same to You,and the  
> arm-none-eabi-g++ 7.2.1

What do you mean about: 
> "cannot replicate"
It's Success or Not Success building it?

kylinpoet, 2018-07-22:
@carlosperate
I used the toolchain as you said..
But When I use yt build ,it shows:
/home/micropython/source/microbit/modos.c:32:10: fatal error: genhdr/microbitversion.h: No such file or directory
 **### #include "genhdr/microbitversion.h"**
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[318/323] Building C object source/CMa...lin/micropython/source/py/modcmath.c.o
ninja: build stopped: subcommand failed.

kylinpoet, 2018-07-22:
> You will need to build using make, that should invoke the script and create the header file. Could you try that and see if it works?
Yes！It's OK now...Thank you for your Helps..

closed 2018-07-22 by kylinpoet

zjuxumang, 2019-03-19:
I meet the same problem , i have tried v1.0.0&v1.0.1 but nothing changed
I competetlly follow the instuctions below:

> Ubuntu users can install the needed packages using:
> 
> sudo add-apt-repository -y ppa:team-gcc-arm-embedded
> sudo add-apt-repository -y ppa:pmiller-opensource/ppa
> sudo apt-get update
> sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord libssl-dev
> pip3 install yotta
> Once all packages are installed, use yotta and the provided Makefile to build. You might need need an Arm Mbed account to complete some of the yotta commands, if so, you could be prompted to create one as a part of the process.
> 
> Use target bbc-microbit-classic-gcc-nosd:
> 
> yt target bbc-microbit-classic-gcc-nosd
> Run yotta update to fetch remote assets:
> 
> yt up
> Start the build:
> 
> make all

dpgeorge, 2019-03-19:
The unused newlib header is now removed, see bf54edfee6b18cd8891f98d82bc1dd3c121e7560