bbcmicrobit/micropython #540: Bugs in NeoPixel in newer firmware versions

rhubarbdog, 2018-07-28:
Hi,
I've bought a Kitronik MOVEmini robot it has 5 multi coloured LED's which are called ZIP LED's, but are compatible with NeoPixel.  The lights fully work using this .hex file (MicroPython v1.7-9-gbe020eb on 2016-04-18) but when I use a more modern firmware which I downloaded in good faith from here ( https://github.com/mcauser/microbit-tm1637/tree/master/firmware ). which has details (MicroPython v1.9.2-34-gd64154c73 on 2017-09-01).

I have 5 neopixel leds on pin0. when i run this program v 1.7 works correctly giving me 5 different coloured lights, but when i use version 1.9.2 all 5 lights are white

[light01.py.txt](https://github.com/bbcmicrobit/micropython/files/2238084/light01.py.txt)

carlosperate, 2018-07-28:
Hi @rhubarbdog,

It's possible this issue was fixed in https://github.com/bbcmicrobit/micropython/pull/513, which has been added to MicroPython after that hex file was uploaded to that repository.

Could you try running your code in http://python.microbit.org/v/beta or the latest version of Mu (https://codewith.mu)?

carlosperate, 2018-07-29:
What other kind of bugs are you seeing in the MicroPython shipped with http://python.microbit.org/v/beta or Mu v1.0.0?
Could you open other issues for each one of those bugs with as much details as you can? Having a different issue per bug will help us keep things organised and easily be able to look into them.

dpgeorge, 2018-07-29:
> hi i've updated both uflash and Mu they are both now flashing firmware 1.9.2-34, but this has got neopixel working. now uflash/mu are up to date i don't need my version of micro python firmware. i required it for i2c.init but it appears to have bugs even though the version details are the same

@carlosperate I think what is being said here is that everything is now working with the very latest firmware.  There is confusion because the custom build with version 1.9.2-34 that had the neopixel bug is now replaced by the very latest version (what we call v1.0.0-rc.2) which still says 1.9.2-34 (for the uPy upstream version).  They needed i2c.init, so needed the custom build, but the custom build had neopixel bugs.  These are now fixed in the very latest v1.0.0-rc.2.

@rhubarbdog can you confirm that everything is now working correctly for you using the very latest firmware, which has v1.0.0-rc.2 written in the version number (after 1.9.2-34)?

rhubarbdog, 2018-07-29:
yes everything ok now. both Mu and uflash now flash firmware 1.9.2-34 v1.0.0-rc.2

carlosperate, 2018-07-29:
Oh, right, apologies for the miscommunication.
Very glad all it is all working now :)

dpgeorge, 2018-07-29:
Thanks for confirming it @rhubarbdog

closed 2018-07-29 by dpgeorge