bbcmicrobit/micropython #544: feature request. button_a.both_are_pressed() / button_b.both_were_pressed()

rhubarbdog, 2018-08-02:
is it possible to add button methods to both buttons a an b to detect when both were pressed. microsoft makecode (java) supports both buttons being pressed as a trigger for a callback. I've written code to execute a function if both are pressed, but it requies loads of running_time() calculations to trigger events a then b or event a+b

rhubarbdog, 2018-08-03:
> Well Java*Script* does things slightly differently (as noted it uses
> events)
>
> both_are_pressed doesn't make sense on a single button
>
> What would this do differently compared to
>
> if button_a.is_pressed() and button_b.is_pressed():
>     # Thing

if you consider the code
```
if button_a.is_pressed() and button_b.is_pressed():
    #item both
elif button_a.is_pressed():
    #item A
elif button_b.is_pressed():
    #item B
```
there's a race condition, it's nearly impossible to press both buttons at
the same time,
I was kind of thinking that when pressing buttons `button_a.is_pressed()`
would return false if both buttons were pressed

`button_a.both_are_pressed()` is a little confusing but the only other
option is to create a new button object `button_ab` or `button_both`

rhubarbdog, 2018-08-03:
if you consider the code 
```
if button_a.is_pressed() and button_b.is_pressed():
     #item both
elif button_a.is_pressed():
    #item A
elif button_b.is_pressed():
#item B 
``` 
there's a race condition, it's nearly impossible to press both buttons at the same time, I was kind of thinking that when pressing buttons `button_a.is_pressed()` would return false if both buttons were pressed.
`button_a.both_are_pressed()` is a little confusing but the only other option is to create a new button object `button_ab` or `button_both`

MrYsLab, 2018-08-03:
If first testing if either of the buttons is pressed and then adding a short delay before reading the other button,  wouldn't that solve the "race" condition?