bbcmicrobit/micropython #545: Documentation: Add compass units for get_x/y/z and get_field_strength

microbit-carlos, 2018-08-20:
The current documentation doesn't list the units for the compass return values: https://microbit-micropython.readthedocs.io/en/latest/compass.html

As MicroPython returns the values reported by the DAL without further processing, then the units in this case are nano teslas: https://lancaster-university.github.io/microbit-docs/ubit/compass/

microbit-rosslowe, 2018-10-04:
A pedantic note but just something to be aware of - the [Tesla](https://en.wikipedia.org/wiki/Tesla_(unit)) is a unit of magnetic field strength and is not equivalent to force (it's Newtons per Amp-metre), perhaps its worth linking to the Wikipedia page to explain what a Tesla is.

Also note, each line states "Gives the reading of the magnetic force on the **`x`** axis, as a positive or negative integer, depending on the direction of the force." whereas it should say x, y, then z.

microbit-carlos, 2018-10-17:
PR #568 has been merged, thanks @microbit-sam and @microbit-rosslowe.

PR #568 (2018-10-17): docs: Add compass units to docs
Magnetic field strength in nano tesla
Closes #545
closed 2018-10-17 by microbit-carlos