bbcmicrobit/micropython #547: The links in  your README.md are incorrect

rhubarbdog, 2018-08-21:
I'm trying to join the mailing list  microbit@python.org (https://mail.python.org/mailman/listinfo/microbit).
when i click on the mailman link the python.org site reports 'no such list microbit'
The informaion i get back from python.org is direct my enquiries to postmaster@python.org when i try this they also report there's no such list.
How do i get involved?

deshipu, 2018-08-21:
That mailing lists has been closed.

rhubarbdog, 2018-08-22:
how do i 'register' to become a contributor?
I can program in python and c, my c++ is at a beginners level to program, but i read it quite well.  Happy to help with testing

deshipu, 2018-08-22:
You don't need to register, simply submit a pull request. If you are not sure about how to solve a particular issue or whether a given feature is desired, create an issue first to discuss it.

Isioviel, 2018-09-17:
I have submitted a pull request to remove the links to the mailing list, let me know if there are any issues! #560

microbit-carlos, 2018-09-18:
Changes have been merged (thanks Damien!) so I'll close this issue.

Thank you @rhubarbdog  for reporting the old link and thank you @Isioviel for submitting PR #560 to update it!

@rhubarbdog if you still have any question about how to contribute to this project please let us know and we can help through the process :)

closed 2018-09-18 by microbit-carlos