bbcmicrobit/micropython #548: Documentation: Info on hex file format

microbit-carlos, 2018-08-22:
It'd be good to document for developers how the hex file is formed, how/where the user code is added into the known flash location, how we are adding the version info into the UICR and its implications in adding user code into the hex.

dpgeorge, 2018-09-03:
Yes, I can write such documentation.  Is it needed for v1.0.0?

microbit-carlos, 2018-09-04:
If you can that would be great, I can probably spend some time early next week to put something together as well.
Doesn't need to be massive, just some general info so that anybody developing an editor or tool knows what's going on . For example, we had to do these two small changes in the Python Editor and uFlash for the UICR addition: https://github.com/bbcmicrobit/PythonEditor/pull/57/commits/5348388acd896320d020ec1c0c25be2fd73e7234
https://github.com/ntoll/uflash/pull/52/files