bbcmicrobit/micropython #551: Translation

franck179, 2018-09-03:
Hi, i'd like to translate the tutorials in french. I've already translated all the .rst files in a forked project but i don't know github very well and i don't know how propose them here and if it's possible to have them online.
Could someone help me with this ?
Thank you

deshipu, 2018-09-03:
I can help you with that. The first step is to have them published on readthedocs as microbit-micropython-fr. Then I can connect it to the default docs so that it appears in the menu.

franck179, 2018-09-03:
does it mean that i should create  a new project called microbit-micropython-fr on readthedoc or does it already exist ?

deshipu, 2018-09-03:
You have to create it. Let me know if you need help with that.

franck179, 2018-09-03:
Ok : once in readthedocs, connected to my github i can import a project. Since I forked the bbcmicro python on my github, shall import this fork to readthedocs ?

deshipu, 2018-09-03:
Yes, precisely.

franck179, 2018-09-03:
Ok, I assume i then choose french as the project language and Python for the coding one.  But what homepage adress shall i write ? https://microbit-micropython.readthedocs.io ?

franck179, 2018-09-03:
Ok I did it :)

deshipu, 2018-09-03:
I guess so. It doesn't really much matter.

deshipu, 2018-09-03:
Hm, I'm sorry, but it seems that now to add a project as a translation I need to be a maintainer on it. Can you add the user "sheep" as a maintainer to your translation on readthedocs? You can do that in Admin->maintainers. Once I add it, you can remove me.

franck179, 2018-09-03:
done

deshipu, 2018-09-03:
Thank you. I have added it, you can see it in the menu at the bottom left of the page that reads "Read the Docs". You can now remove me as a maintainer, if you want.

franck179, 2018-09-03:
Thank you very much for your help :-D

franck179, 2018-09-04:
One last thing .... (;-) )
When on this page : https://www.microbit.org/fr/guide/python/ which is writtent in french (as one can guess it by reading the url !) all the links point to the https://microbit-micropython.readthedocs.io/en/latest/tutorials/ pages instead of https://microbit-micropython.readthedocs.io/fr/latest/tutorials/ ones which would make more sense.
Is it possible to change this behaviour ?

microbit-carlos, 2018-09-04:
Unfortunately this is not something we could do at the moment.

Converting a microbit.org link into a localised link (so that a link in the `fr` translated page has a URL pointing to the `fr` version) would dynamically convert the links for all the available translations, as it's one of those "all or nothing" features.
And because there are more translations in microbit.org than the MicroPython documentation we would end up with broken links in all those translations.