bbcmicrobit/micropython #556: Using Microbit pins 5 and 11 as inputs gets Error pin in button mode

4tronix, 2018-09-09:
Bit:Bot uses pins 5 and 11 as digital inputs and all existing code simply calls the read digital functions. This now gives an error stating the pin is in button mode, but it has worked fine for the last 2 years. How do we change pin modes? I can’t find documentation for it anywhere

DavidWhaleMEF, 2018-09-09:
@carlosperate @dpgeorge any ideas? If you could point us at the general area or change that caused this, I can probably bake up a workaround for @4tronix so that their line following buggy can be made to work again, thanks!

DavidWhaleMEF, 2018-09-09:
@4tronix Just a side note, we get a number of tickets logged about bit:bot keep going into pairing mode. This is because pairing mode is entered by holding buttons A and B when powering up or resetting. So if the bit:bot is in a state where both inputs are active on power up (presumably on the floor) and a .hex from MakeCode has been loaded (which has the Bluetooth DFU present), it enters the Bluetooth pairing mode app.

If not done already, it might be worth you signposting this in your docs? Thanks!

4tronix, 2018-09-09:
Yes, it is in big red letters at the top of 4tronix.co.uk/bitbot
For the last iteration (v1.2) I have also delayed the activation of the sensors for 2 seconds to prevent this.

Best regards,
Gareth

dpgeorge, 2018-09-10:
> This now gives an error stating the pin is in button mode, but it has worked fine for the last 2 years

Actually, it was about 2 years ago that this was changed!  See: ce36e0dea1c23f0608493978c839707386f34ce6

So it's strange that this issue only just came up now.

> How do we change pin modes?

It's not possible, at least not for pins attached to buttons.  The only way to read the current digital value of a button is to use `button_a.is_pressed()`.

---

Probably this behaviour should be changed back to the old behaviour that did allow to read a button like a digital IO pin.  I made a PR for that, see #557

4tronix, 2018-09-10:
> <mention@noreply.github.com>
> inputs gets Error pin in button mode (#556)

I guess it is because this version has only now been incorporated into a 
released version of mu editor which people use.

It would be good if the behaviour could be changed back so that the 
existing line follower code can continue working.

In the meantime I will tell customers to convert to 
button_a.is_pressed() usage.

microbit-carlos, 2018-09-10:
Thanks Damien for the quick turnaround on the fix. The PR adds the ability to use `read_digital` (and `set/get_pull`) on a pin in button mode.
We'll include this update in the next MicroPython RC and we are getting very close to a final v1.0.0.

DavidWhaleMEF, 2018-09-10:
Remember that the micro:bit hardware has external pull-up resistors fitted to both buttons, so the pull effects might operate differently on these pins due to interaction between the internal pullups and the external hard pullups. We signpost this in the tech.microbit.org docs, but it would be worth checking that the readthedocs for micropython also alerts users to this.

microbit-carlos, 2018-09-10:
Good point @DavidWhaleMEF, I've opened this issue to update the docs: https://microbit-micropython.readthedocs.io/en/latest/pin.html

dpgeorge, 2018-09-10:
This particular issue is now fixed by 3eaa746f98a07e09763a61cc707251020525fa4a

closed 2018-09-10 by dpgeorge

DavidWhaleMEF, 2018-09-10:
@carlosperate @dpgeorge Is there a cached .hex file somewhere of this new build (and standard instructions on how to override a .hex baked in to the IDE) that @4tronix can try, to confirm it fixes the problem on their specific platform? 

The problem (as always) with closing the ticket as 'fixed in master' is it still could be some time before the customer that reported the issue can actually check that it works for them via standard routes, and we can't expect them to have the time to build everything from source, like you can.

Thanks!

microbit-carlos, 2018-09-10:
There is no cached hex file yet, we are working towards releasing an RC4 very soon and we'll tag here and update the online editor at the same time.

We'll probably only upload a hex file to a GH release when the final v1.0.0 is released. But I can add a hex here for convenience once we have RC4.

Instruction on how to point to a different hex in Mu can be found in the Mu website: https://codewith.mu/en/howto/microbit_settings

rhubarbdog, 2018-12-20:
Hi
I'm planning to create a .hex file once per month and host them on github to allow people at the edge of microbit the chance to get their code working (again).
This button mod is in <a href="http://github.com/rhubarbdog/microbit-beta">microptython-20181219.hex</a>