bbcmicrobit/micropython #565: Outdated reference to TouchDevelop on "Next Steps" docs page

ScottDWebster, 2018-09-19:
The sentence in question (from https://microbit-micropython.readthedocs.io/en/latest/tutorials/next.html):
"To encourage you we have hidden a number of Easter eggs in MicroPython and the code editors (both TouchDevelop and Mu)."
There seems to no longer be a TouchDevelop editor for the Micro:bit.  https://www.touchdevelop.com/ links to the MakeCode editor.  I have no knowledge of what Easter eggs might exist in the other editors so I am unsure how the sentence ought to be rewritten.  There would be at least three editors: (1) https://python.microbit.org/v/1, (2) https://makecode.microbit.org/ and (3) Mu,

jaustin, 2018-10-03:
I think the sentence should probably just remove the TD link
I've made the change in a docs branch here
https://github.com/bbcmicrobit/micropython/tree/jaustin-docs

I'll queue a few more docs changes before I PR that, but do you think the change https://github.com/bbcmicrobit/micropython/commit/7cd616018ed050908c23afb91b232abf6a749c82 does the trick?

microbit-carlos, 2018-10-09:
@jaustin I think it's preferable to have individual PRs per doc change, so unless there are other updates needed to the "next steps" page I would just submit that as an individual PR.

microbit-carlos, 2018-10-16:
PR submitted: https://github.com/bbcmicrobit/micropython/pull/577

microbit-carlos, 2018-10-17:
#577 has been merged to master. Thanks @ScottDWebster and @jaustin!

closed 2018-10-17 by microbit-carlos