bbcmicrobit/micropython #566: feature request - Servo control

rhubarbdog, 2018-09-21:
Hi,
I'm getting into robots for the microbit. One popular solution is to use continuous servos to control wheels and standard angular servos for robot arms.
I have written a servo class which is now an 870 byte python script. There's then the robot class and finally the program. The problem is i'm out of memory and the robot i have has additional sensors and features which can be bought. My robot is a white elephant from a micropython point of view.

Microsoft blocks has a write_servo block which has 2 parameters a pin and an angle between 0 and 180.
Lancaster university's `microbit.h` for c++ programming has a `setServoValue` and a `setServoPulseUs` method for pins.

Microsoft blocks and Lancaster university's c++ header have different ideas about what if any circuitry is required. Blocks uses direct wiring to microbit or a circuit equivalent to 
[image]
where as Lancaster university has an inverted logic and a servo must be connected to a microbit as follows 
[image]

What's the implications of adding the the following:
`pinX.write_servo()`
`pinX.configure_servo()` 

I believe that both Microsoft Blocks and Lancaster University are correct in there circuit requirement. I have the need for Microsoft logic but implementing both isn't rocket science.

deshipu, 2018-09-21:
All you need to control a servo is this:
```python
pin.set_analog_period(20)
pin.write_analog(int(25 + 100 * degrees / 180))
```
There is no need for dedicated methods or libraries.

rhubarbdog, 2018-11-29:
thanks that works, i've tweaked the values in the `write_anaolg(int(25 + 85 * degrees / 180))` it gave a better response to the input ange

closed 2018-11-29 by rhubarbdog