bbcmicrobit/micropython #567: Add v1.0 features to documentation

microbit-rosslowe, 2018-10-03:
There are a number of new features in MicroPython v1.0 that need documenting.

Changelog, for reference: https://gist.github.com/microbit-carlos/0b116badf3bafe7100ce7e7a1a39ac86

- [ ] Add `get_pull()`, `get_mode()` to pins documentation https://microbit-micropython.readthedocs.io/en/latest/pin.html#microbit.MicroBitTouchPin -> page change
- [ ] replace `uart.readall()` with `uart.read()` https://microbit-micropython.readthedocs.io/en/latest/uart.html?highlight=readall -> page change
- [ ] documentation on the 'machine' module -> new page
- [ ] add documentation on `help('modules')` -> is there a page this can be added to?
- [ ] add notes on `time` and `utime` modules -> new page(s)
- [ ] `micropython.const()`,  `micropython.opt_level()` and `micropython.kbd_intr()` (all need documenting, http://docs.micropython.org/en/v1.9.2/pyboard/library/micropython.html) -> add to page

microbit-rosslowe, 2018-10-03:
@microbit-carlos

microbit-rosslowe, 2018-10-04:
`time` / `utime`:
```
  sleep -- <function>
  sleep_ms -- <function>
  sleep_us -- <function>
  ticks_ms -- <function>
  ticks_us -- <function>
  ticks_add -- <function>
  ticks_diff -- <function>
```

`machine`
```
  unique_id -- <function>
  reset -- <function>
  freq -- <function>
  disable_irq -- <function>
  enable_irq -- <function>
  mem8 -- <8-bit memory>
  mem16 -- <16-bit memory>
  mem32 -- <32-bit memory>
  time_pulse_us -- <function>
```

microbit-rosslowe, 2018-10-09:
@microbit-carlos Thanks, I'll take a look at this today.

 Is `utime` just a MicroPython port of `time`, and `import time` is the same as `import utime` for the sake of continuity?

What do `get_pull()`, `get_mode()` do?

microbit-carlos, 2018-10-11:
> Is utime just a MicroPython port of time, and import time is the same as import utime for the sake of continuity?

Yes, that correct, in fact you can try this on the REPL:

```
MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit v1.0.0 with nRF51822
Type "help()" for more information.
>>> import utime
>>> import time
>>> time
<module 'utime'>
>>> time == utime
True
```

> What do get_pull(), get_mode() do?

The `get_mode()` method on a pin will return the pin mode. When a pin is used for a specific function, like writing a digital value, or reading an analog value, the pin mode changes. The current pin modes can be seen here: https://github.com/bbcmicrobit/micropython/blob/3eaa746f98a07e09763a61cc707251020525fa4a/source/microbit/microbitpinmode.c#L97-L109

The `get_pull()` method returns the pull configuration on a pin, which can be one of 3 values `NO_PULL`, `PULL_DOWN`, or `PULL_DOWN`. These value can be manually set using the `set_pull()` method, e.g. `pin12.set_pull(pin12.PULL_DOWN)`, and it will also be automatically configured when a pin mode requires it. For example, when `read_digital()` is used for the first (or the first time used after the pin is configured in a different mode)  it will configure that pin pull up down.

> Is there somewhere help('modules') can be added to? Possibly https://microbit-micropython.readthedocs.io/en/latest/microbit_micropython_api.html

Yes, that looks like the best place to me.

I still have a question left, I'll respond that one later today.

microbit-rosslowe, 2018-10-11:
I've had a go at adding all new MicroPython 1.0 features to the docs; please see the PR above.

closed 2018-12-12 by microbit-carlos

microbit-carlos, 2020-08-24:
I'm not sure why, but the "latest" tag of readthedocs wasn't being updated on each commit on master (as suggested by the settings), but I have manually triggered a new build and updated the settings to see if it better tracks master now.

It might take a little while for the website to show the latest version, but in the meantime you can see the documentation with the GitHub viewer: https://github.com/bbcmicrobit/micropython/blob/master/docs/pin.rst#classes

kevinjwalters, 2020-08-24:
@microbit-carlos Thanks. I put a question in the forum about a finer point on `get_mode()`: [get_mode for micro:bit pins with analog read and write](https://forum.micropython.org/viewtopic.php?f=2&t=8933).

PR #578 (2018-12-12): docs: Add MicroPython 1.0 Features (#567)
Update docs to include info on v1.0 features, closes #567

- remove 1.0 notice from microbit_micropython_api.rst
- remove "WARNING, THIS IS ALPHA CODE"... from speech.rst
- replace uart.readall() with uart.read() in microbit_micropython_api.rst
- replace uart.readall() with uart.read() in uart.rst
- create micropython.rst, listing new features in the micropython module, and add a reference in index.rst (source: http://docs.micropython.org/en/v1.9.2/pyboard/library/micropython.html)
- create utime.rst, listing new features in the utime module, and add a [... cut]