bbcmicrobit/micropython #591: readall method does not exist

DavidWhaleMEF, 2018-10-30:
The currently published readthedocs for the uart module refer to a 'readall' method.

https://microbit-micropython.readthedocs.io/en/latest/uart.html

[image]

The current v1.0.0 MicroPython image does not have this method:

```
MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit v1.0.0 with nRF51822
Type "help()" for more information.
>>> dir(uart)
['init', 'any', 'read', 'readline', 'readinto', 'write', 'ODD', 'EVEN']
```

DavidWhaleMEF, 2018-10-30:
Note, it looks like this functionality can be achieved with readinto(), the main difference being that you provide the buffer, rather than having the buffer created for you (as the old readall did).

I recall reading that @dpgeorge had removed this a while ago, but it's still in the live docs, and the live docs don't match the live language (I appreciate we're all working to correct that at the moment though!)

DavidWhaleMEF, 2018-10-30:
Note: Removing methods will break existing users code. We all know that, but its worth remembering that, as it provides a bad experience for users who expect their code to always work on the latest editors and languages.

Should we mark removed methods in the docs as 'deprecated, please use x instead', and have a deprecation process a bit like Java, where one major release marks it as deprecated, the next major release actually removes it?

Comments welcomed @carlosperate @jaustin

microbit-carlos, 2018-10-31:
Completely agree that we should have a good deprecation process, although the aim is to never have to use it.

The `readall` method was removed in https://github.com/bbcmicrobit/micropython/commit/d231463dd9a4c9fe366314c2a59d5ff507a22893, which was over 2 years ago. At this point I think we should add a note to the uart docs to indicate it has been deprecated. I've created this PR, let me know what you think:  https://github.com/bbcmicrobit/micropython/pull/595

As far as achieving the same functionality goes, you can use the `read()` method without any arguments to do the same. PR https://github.com/bbcmicrobit/micropython/pull/588 clarifies the method behaviour if an argument is not provided.

microbit-carlos, 2018-11-05:
Thanks David, the docs have been updated in PR #595.

closed 2018-11-05 by microbit-carlos