bbcmicrobit/micropython #593: uart.readinto returns result different to that documented

DavidWhaleMEF, 2018-10-30:
```
>>> buf = bytearray()
>>> buf
bytearray(b'')
>>> uart.readinto(buf)
0
>>> buf
bytearray(b'')
>>> 
```

Result printed (no uart activity) is 0. Docs say it will be None on timeout?

https://microbit-micropython.readthedocs.io/en/latest/uart.html

[image]

microbit-carlos, 2018-10-31:
I guess technically the number of bytes read was 0, but you are right, reading that I'd have thought  it would time-out and return None.

The same thing happens even if nbytes is specified:

```
>>> buf = bytearray()
>>> uart.readinto(buf, 3)
0
```

@dpgeorge When can we expect the return value be `0` vs `None`?

dpgeorge, 2018-11-01:
The behaviour of `readinto` is supposed to match CPython, see https://docs.python.org/3/library/io.html#io.RawIOBase.readinto.  And I'm pretty sure it does match that definition, and the behaviour is also matching the microbit docs here.

I think the gotcha is that doing `buf=bytearray()` creates a zero-length bytearray, and so a request to read zero bytes is always satisfied, and 0 is returned.

> The same thing happens even if nbytes is specified:

You can't specify more bytes than are in the supplied buffer, in this case zero.

microbit-carlos, 2018-11-02:
Thanks Damien, that makes sense:

```
>>> buf = bytearray(5)
>>> print(uart.readinto(buf))   
None
>>> print(uart.readinto(buf, 5))
None
```

I don't think there needs to be any action on the docs, so I'll close the issue. @DavidWhaleMEF if you think otherwise feel free to reopen.

closed 2018-11-02 by microbit-carlos