bbcmicrobit/micropython #594: uart.read does not document return results

DavidWhaleMEF, 2018-10-30:
The current readthedocs does not say what data type and value uart.read() returns

https://microbit-micropython.readthedocs.io/en/latest/uart.html

[image]

It says:
```
Read characters. If nbytes is specified then read at most that many bytes.
```

So, does it read characters or does it read bytes? What does it return? characters does not equal bytes where character encodings are used. Also, what is the return type? If it returns characters, is this a string? If it returns bytes, is it a bytes() or a bytearray()?

The implication is that if you don't provide a parameter it reads characters, and if you do provide a parameter, it reads bytes, is that correct?

DavidWhaleMEF, 2018-10-30:
By experimentation alone, the return type appears to be a bytes() object.

It didn't seem clear to me from the documentation that this was the case, because it talks about characters (which implies a string) and bytes (which could be a bytes() or a bytearray() )

microbit-carlos, 2018-10-31:
PR https://github.com/bbcmicrobit/micropython/pull/588 already covered the return type, was that enough or does it need something more explicit?

DavidWhaleMEF, 2018-10-31:
Thanks, commented in the PR. I will close this issue.

closed 2018-10-31 by DavidWhaleMEF