bbcmicrobit/micropython #599: feature request - pulseIn

Monsoon117, 2018-11-21:
Hi, I try to use Ultrasonic sensor on Micobit.
I found I can't do that without PluesIn() which can return the length of the pulse.
Is it possible to add this feature?

Monsoon117, 2018-11-26:
> 
> 
> You can try `machine.time_pulse_us()` which is documented here: http://docs.micropython.org/en/latest/library/machine.html?highlight=time_pulse_us#machine.time_pulse_us

Thanks for replys. Micro:bit's dont have machine.time_pulse_us(). I try to use running_time() to replace it.
But it don't work. The disatance returned is a fixed value.

`def distance():
    distance = 0

    pin8.write_digital(1)
    sleep(10)
    pin8.write_digital(0)

    while pin1.read_digital()==0 :
        pass
    if pin1.read_digital() == 1:
        t1 = running_time()
        while pin1.read_digital()==1 :
            pass
        t2 = running_time()
        print(t2)
        print(t1)
        distance = (t2-t1)*340/10000
    return (distance)`

Monsoon117, 2018-11-26:
> 
> 
> Hi Monsoon117,
> Version 1.0.0 contains the `machine` module and `time_pulse_us` function. How are you programming your micro:bit with MicroPython?

I use Pycharm with MicroPython Plugin. Thank you so much. I found them when I use Mu-editor.

dpgeorge, 2018-11-27:
It sounds like it's working now, so closing this issue.

closed 2018-11-27 by dpgeorge