bbcmicrobit/micropython #600: neopixel effects music

rhubarbdog, 2018-11-21:
i've bought <a href="https://www.kitronik.co.uk/5626-game-zip-64-for-the-bbc-microbit.html">this joypad (kitronik game zip)</a>.
It has a little speaker and an array of "zip LEDs" (neopixel compatible addressable LEDs)
when i run the following program the music plays once at the start and then about every 5 loops randomly.
```
from microbit import *
import music
import neopixel

zip_led = neopixel.NeoPixel(pin0, 64)

while True:
    display.show('-')
    music.play(music.BA_DING, pin2, False)
    display.clear()

    sleep(1000 * 1)
    
    zip_led.clear()
    zip_led[1] = (20, 20, 0)
    zip_led.show()

    sleep(1000 * 2)
```

dpgeorge, 2018-11-22:
From the code above, it looks like the music tune is playing in the background but finishing (via sleep(1000*1)) before the LEDs are written to.  So there shouldn't be any interference between the music and LEDs.  What happens if the music plays in the foreground, does it work then?

deshipu, 2018-11-22:
Out of curiosity, how are you powering it? Perhaps it's restarting because of the voltage sag due to the neopixels lighting up? Do you get anything displayed on the serial?

rhubarbdog, 2018-11-22:
I'm powering it of 3 * 1.5volt alkaline batteries. Which are new, but not heavy duty type like duracell.  The microbit is powered by theses via the GPIO connector. I've tried running the code from the initial post with a usb power lead in to keep the microbit alive the music still stops after 1 play and then only plays on some loops, it appears to be more like 1:11.
You ask "Do you get anything displayed on the serial?", how would i test this? i have just tried adding a couple of print statements and checked they were working using a REPL session with `screen`

rhubarbdog, 2018-11-23:
I've tried the following
1) using a different speaker, some genuine neopixel (ws2812 addressable LEDs) and a different microbit i ran the code in my original post. I have used a level shifter between the microbit and neopixels which have their own 5volt supply. Music playback is still defective i.e. plays the first time, then 1:6 -> 1:11 playbacks:loops
2) to isolate this further i produced <a href="https://makecode.microbit.org/_JDmU0UTxm39E">these blocks</a> and music plays correctly every loop

dpgeorge, 2018-11-27:
I can confirm this issue.  The problem is that when there are lots of neopixels the interrupts are disabled for a long time, and the PWM (which drives the music) stops functioning.  This is because it is software based, misses an interrupt, and the next time it gets a chance to service an interrupt and update the counter for the time of the next interrupt, that next interrupt already passed.  In that case it has to wait for the counter to wrap around before it starts functioning again.  Since the counter is 24 bits and running at 1MHz, it takes about 16seconds to start working again.

rhubarbdog, 2018-11-27:
Just so you're aware. I was able to reproduce the bug with only 8 neopixels

rhubarbdog, 2020-05-12:
I would like to get going with the joypad i bought i'm thinking about writing an article for it for micro:mag but need a .hex file that it would work with. has the relevant pull request been joined to a repository so i can test the fix and hopefully write a good game to show off what a cool daughter board this is for the microbit

PR #607 (2021-11-29): lib: Make sure that PWM continues to work even if it misses an IRQ.
If the soft PWM driver misses an IRQ then the next CC value (computed as CC[x] += delay) could already be behind the timer counter value, and hence the next compare IRQ won't fire until the timer counter value wraps around.

This needs some testing because the timing of the PWM will now be slightly different compared to before, since the next event will be computed from the *current counter value* plus delay, rather than the counter value at the *time of the previous IRQ*.  But I can't really think of any other way to fix the  [... cut]
closed 2021-11-29 by dpgeorge