bbcmicrobit/micropython #603: Image aliases for ROCK and PAPER

microbit-giles, 2018-11-22:
Can we add ‘aliases’ so Image.ROCK produces the same image as Image.SQUARE_SMALL and Image.PAPER was the same as Image.SQUARE ?

This would further simplify making a rock, paper, scissors game and make MicroPython even more accessible to more young people.

dpgeorge, 2018-11-27:
The cost (in code space) to add these aliases is about 12 bytes per alias.  So not that much.  On the other hand, it's not so hard to explain that SQUARE looks like paper so that's why it can be used to make the game.

microbit-giles, 2018-11-27:
That's a very fair point @dpgeorge! Probably not worth pursuing unless anyone else thinks it's a good idea.

dpgeorge, 2018-12-02:
> I'd love that free memory for my programs.

Note that adding the alias takes 12 bytes of *flash* space, not RAM.

rhubarbdog, 2018-12-02:
Would freeing flash increase the 8188 byte script size limit `uflash`reports. Any extra space for user code would be greatfully recieved.  The microbit has a bit of a vic 20 feel. Looks like a Commodore, feels like a Commodore,  but ain't a C64. Some of the smaller machines available when i were a boy shouldn't have existed.

dpgeorge, 2018-12-03:
> Would freeing flash increase the 8188 byte script size limit `uflash`reports.

No, that size is fixed.  Instead you can put your code on the device's internal filesystem as main.py (and also put libraries in separate files), then you have more space to use.  Mu supports this feature now.

dpgeorge, 2021-09-06:
Note that SCISSORS was added in 5005880aa3acd5e1fde9b09b9de85932e65353f1

microbit-carlos, 2022-03-25:
I think this can be closed now, thanks everyone!

closed 2022-03-25 by microbit-carlos