bbcmicrobit/micropython #604: 404 not found [IP:91.189.95.83] while sudo add-apt-repository -y ppa:pmiller-opensource/ppa

jxgx072037, 2018-11-26:
Hi，

I met an error while I run
`sudo add-apt-repository -y ppa:pmiller-opensource/ppa`

I want to install the microbit-micropython on ubuntu 18.04.1 LTS

Here is the error report:

```
Ign:1 http://mirrors.aliyun.com/ubuntu trusty InRelease
Hit:2 http://mirrors.aliyun.com/ubuntu trusty-security InRelease
Hit:3 http://mirrors.aliyun.com/ubuntu trusty-updates InRelease                                 
Hit:4 http://mirrors.aliyun.com/ubuntu trusty-proposed InRelease                                
Ign:5 http://ppa.launchpad.net/pmiller-opensource/ppa/ubuntu bionic InRelease                   
Hit:6 http://mirrors.aliyun.com/ubuntu trusty-backports InRelease                               
Hit:7 http://mirrors.aliyun.com/ubuntu trusty Release                                           
Hit:9 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu bionic InRelease                
Err:10 http://ppa.launchpad.net/pmiller-opensource/ppa/ubuntu bionic Release  
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/pmiller-opensource/ppa/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Is there anyway to fix this?

carlosperate, 2018-11-26:
Hi @jxgx072037,

Did you also add the `team-gcc-arm-embedded/ppa` PPA?

jxgx072037, 2018-11-27:
> Hi @jxgx072037,
> 
> Did you also add the `team-gcc-arm-embedded/ppa` PPA?

Yes, I could add the team-gcc-arm-embedded/ppa successfully

carlosperate, 2018-11-27:
And did you do a `sudo apt-get update` after adding those two PPAs and try to install it again? The team-gcc-arm-embedded group has images for 18.04, so it should work on bionic.

carlosperate, 2018-11-28:
`gcc-arm-embedded` should be installed from the `team-gcc-arm-embedded` PPA: https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa?field.series_filter=bionic

I don't think you need `pmiller-opensource` any more.

carlosperate, 2018-12-12:
Hi @jxgx072037, I was just wondering if you still had issues installing gcc in 16.04?