bbcmicrobit/micropython #605: Compiling Problem

lego3708, 2018-11-26:
Hi.
I have just managed to compile the firmware on a Linux VM, but when I set it as my firmware in Mu, and try to flash a hello world program, It gives an error. Here's the log
```
2018-11-26 19:30:03,154 - mu.logic:645(restore_session) INFO: Loaded files.
2018-11-26 19:30:03,154 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-11-26 19:30:03,154 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False
2018-11-26 19:30:03,154 - mu.logic:658(restore_session) INFO: Custom micro:bit runtime path: D:\Documents\Programming\Python\Microbit\custom\firmware.hex
2018-11-26 19:30:03,176 - mu.logic:1112(change_mode) INFO: Workspace directory: C:\Users\l3gom\mu_code
2018-11-26 19:30:04,966 - mu.logic:695(new) INFO: Added a new tab.
2018-11-26 19:30:22,033 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2018-11-26 19:30:22,033 - mu.modes.microbit:249(flash) DEBUG: Python script:
2018-11-26 19:30:22,033 - mu.modes.microbit:250(flash) DEBUG: b'from microbit import *\r\ndisplay.scroll("Hello World!")'
2018-11-26 19:30:22,034 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: F:\
2018-11-26 19:30:22,035 - mu.modes.base:217(find_device) INFO: Found device on port: COM4
2018-11-26 19:30:22,035 - mu.modes.base:218(find_device) INFO: Serial number: 9900000040554E450039400C0000003F0000000097969901
2018-11-26 19:30:22,035 - mu.modes.microbit:302(flash) INFO: Serial port: COM4
2018-11-26 19:30:22,035 - mu.modes.microbit:303(flash) INFO: Device serial number: 9900000040554E450039400C0000003F0000000097969901
2018-11-26 19:30:22,035 - mu.modes.microbit:328(flash) INFO: Checking target device.
2018-11-26 19:30:22,539 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0', 'version': 'micro:bit v1.0.0+unknown on 2018-11-26; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2018-11-26 19:30:22,539 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 1.0.0
2018-11-26 19:30:22,539 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.0
2018-11-26 19:30:22,539 - mu.modes.microbit:381(flash) INFO: Flashing new MicroPython runtime onto device
2018-11-26 19:30:26,589 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\l3gom\AppData\Local\python\mu\Logs\mu.log
2018-11-26 19:30:31,719 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2018-11-26 19:30:31,719 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\r\\ndisplay.scroll("Hello World!")\')', 'fd.close()']
2018-11-26 19:30:32,835 - mu.modes.microbit:532(flash_failed) ERROR: Could not enter raw REPL.
2018-11-26 19:30:32,850 - mu.interface.main:711(show_message) DEBUG: There was a problem flashing the micro:bit.
2018-11-26 19:30:32,850 - mu.interface.main:712(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2018-11-26 19:30:34,498 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\l3gom\AppData\Local\python\mu\Logs\mu.log
2018-11-26 19:30:43,133 - mu.interface.main:251(get_save_path) DEBUG: Getting save path: C:/Users/l3gom/mu_code/hi
2018-11-26 19:30:43,133 - mu.logic:876(check_for_shadow_module) INFO: Checking path "C:/Users/l3gom/mu_code/hi" for shadow module.
2018-11-26 19:30:43,133 - mu.logic:842(save_tab_to_file) INFO: Saving script to: C:/Users/l3gom/mu_code/hi.py
2018-11-26 19:30:43,133 - mu.logic:843(save_tab_to_file) DEBUG: from microbit import *

display.scroll("Hello World!")
2018-11-26 19:30:45,283 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2018-11-26 19:30:45,283 - mu.modes.microbit:249(flash) DEBUG: Python script:
2018-11-26 19:30:45,283 - mu.modes.microbit:250(flash) DEBUG: b'from microbit import *\r\ndisplay.scroll("Hello World!")'
2018-11-26 19:30:45,284 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: F:\
2018-11-26 19:30:45,285 - mu.modes.base:217(find_device) INFO: Found device on port: COM4
2018-11-26 19:30:45,285 - mu.modes.base:218(find_device) INFO: Serial number: 9900000040554E450039400C0000003F0000000097969901
2018-11-26 19:30:45,285 - mu.modes.microbit:302(flash) INFO: Serial port: COM4
2018-11-26 19:30:45,285 - mu.modes.microbit:303(flash) INFO: Device serial number: 9900000040554E450039400C0000003F0000000097969901
2018-11-26 19:30:45,285 - mu.modes.microbit:328(flash) INFO: Checking target device.
2018-11-26 19:30:45,788 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0', 'version': 'micro:bit v1.0.0+unknown on 2018-11-26; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2018-11-26 19:30:45,788 - mu.modes.microbit:351(flash) INFO: Board MicroPython: 1.0.0
2018-11-26 19:30:45,788 - mu.modes.microbit:353(flash) INFO: Mu MicroPython: 1.0.0
2018-11-26 19:30:45,788 - mu.modes.microbit:381(flash) INFO: Flashing new MicroPython runtime onto device
2018-11-26 19:30:54,970 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2018-11-26 19:30:54,970 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\r\\ndisplay.scroll("Hello World!")\')', 'fd.close()']
2018-11-26 19:30:56,085 - mu.modes.microbit:532(flash_failed) ERROR: Could not enter raw REPL.
2018-11-26 19:30:56,086 - mu.interface.main:711(show_message) DEBUG: There was a problem flashing the micro:bit.
2018-11-26 19:30:56,086 - mu.interface.main:712(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2018-11-26 19:30:57,361 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\l3gom\AppData\Local\python\mu\Logs\mu.log
```

dpgeorge, 2018-11-27:
> 2018-11-26 19:30:56,085 - mu.modes.microbit:532(flash_failed) ERROR: Could not enter raw REPL.

This line means there is something wrong with the USB serial connection to the micro:bit.  To debug further you'll need to try and access the REPL manually, using eg `picocom /dev/ttyACM0 -b115200`

carlosperate, 2018-11-27:
Mu is able to read the MicroPython version info, so it can access the MicroPython serial, as it has to enter the raw REPL and parse the result from `os.uname()` to get this info:

```
2018-11-26 19:30:45,285 - mu.modes.microbit:328(flash) INFO: Checking target device.
2018-11-26 19:30:45,788 - mu.modes.microbit:332(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.0', 'version': 'micro:bit v1.0.0+unknown on 2018-11-26; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
```

This sounds like an issue within Mu, could you open a GH issue in that repository? https://github.com/mu-editor/mu/issues/

lego3708, 2018-11-27:
Yeah, that sounds like a Mu issue, so I'll do that. Just one more thing though, I followed the firmware flash instructions on: https://microbit.org/guide/firmware/ and the microbit produced FAIL.txt, saying the file wasn't compatible. Is this just because it doesn't support unofficial firmware?

carlosperate, 2018-11-27:
Were you trying to load a DAPLink hex file (0250_kl26z_microbit_0x8000.hex) in Maintenance mode, or a MicroPython hex file?

The micro:bit firmware in that article refers to the software running in a secondary microcontroller (KL26), which gives you the USB storage device for flashing the micro:bit main microcontroller (nRF51).

There are no restrictions on the micro:bit loading alternative software. For instance, you can load J-Link as the firmware (https://www.segger.com/products/debug-probes/j-link/models/other-j-links/bbc-microbit-j-link-upgrade/) in the KL26, or load any hex file in the nRF51 (as long as the hex files doesn't try to write to memory locations not available in the microcontroller).

lego3708, 2018-11-28:
My main problem was trying to make room for a larger program on the microbit, so if anyone has any suggestions for that, or compiling this repo properly, then I would be very grateful.

lego3708, 2018-11-28:
Ah, don't worry. I see where I went wrong. I thought the micropython firmware was a replacement for the DAPlink firmware, but you have to use makecombinedhex.py first. Thanks guys

carlosperate, 2018-11-28:
The firmware from https://microbit.org/guide/firmware/  is not related to MicroPython. The MicroPython hex files are flashed like any other micro:bit program and there is no restrictions.

How large was your Python program? Was the problem running out of RAM or flash?

lego3708, 2018-11-29:
The original issue was running out of flash, but I fixed that and now I'm running out of RAM!

carlosperate, 2018-12-12:
Hi @std, what problems are you seeing? I've been building MicroPython on 16.04 VMs using Vagrant and these scripts: https://github.com/carlosperate/microbit-dev-env/

There is also these two issues more related than this one, where we can continue this thread: https://github.com/bbcmicrobit/micropython/issues/604 and https://github.com/bbcmicrobit/micropython/issues/619