bbcmicrobit/micropython #608: importError

rogasb, 2018-11-28:
Using:
Python 3.7.1 (v3.7.1:260ec2c36a) [MSC v.1915 32 bit (Intel)] on win32. 

the both script and interactive command 'import music' produced the following.

(this error report from interactive command in PyScipter)
 File "C:\Users\Roger\AppData\Local\Programs\Python\Python37-32\lib\site-packages\music\__init__.py", line 2, in <module>
    from . import utils, tables, synths, effects, structures, singing
ImportError: cannot import name 'structures' from 'music' (C:\Users\Roger\AppData\Local\Programs\Python\Python37-32\lib\site-packages\music\__init__.py)

module music was loaded today(28/11/2018), via pip, and then an upgrade was attempted (again via pip)
inspection of the folder shows no file titled 'structures'

At my stage of learning the language I am totally stopped.

rhubarbdog, 2018-11-29:
`import music` is a microbit micropython module.
you should not be using `python3` as a command to run it
1) plug a microbit into your computer
2 a) using mu which can be installed from <a href="https://codewith.mu/en/download">here</a>
  b) using an editor of your choice and <a href="https://github.com/ntoll/uflash">uflash, which lives here<a>
 c) code it live in a REPL session, <a href="https://microbit-micropython.readthedocs.io/en/latest/devguide/repl.html">instructions are here.<a>
 d) use a web based editor <a href="https://microbit.org/code/">at microbit.org</a>. click download to convert your program to a `.hex` file. this can be copied to the microbits flash with file manager.
3) try the examples on <a href="https://microbit-micropython.readthedocs.io/en/latest/">readthedocs.io</a>. navigate the site with the index on the right this will take you straight to<a href="https://microbit-micropython.readthedocs.io/en/latest/tutorials/music.html">music</a> <a href="https://microbit-micropython.readthedocs.io/en/latest/music.html">further reference here</a>

rogasb, 2018-11-29:
Thank you for your very comprehensive answer.

In my naivety I had assumed that all python development was intended to be multi-platform

closed 2018-11-29 by rogasb