bbcmicrobit/micropython #613: OSError when overwriting a large file with a large file

rhubarbdog, 2018-12-06:
I'm planning a project which uses a large file so i thought i'd try some dummy runs before embarking on my program. I've written this program `test_fs.py` it writes 160 lines of 128 characters 20kb i was hoping.#
when i run it on a microbit which already has the file `20k.txt` on it i get the error
OSError: 28

[test_fs.py.txt](https://github.com/bbcmicrobit/micropython/files/2651808/test_fs.py.txt)

may be releated to <a href="https://github.com/ntoll/microfs/issues/12">ntoll/microfs/#12</a>

dpgeorge, 2018-12-07:
> OSError: 28

This means ENOSPC, no space left on device.

But looking at your code, it opens the file in mode "w" which should trancate/free any existing file by this name.  So in principle the code should work without raising the ENOSPC error.  This requires more investigation.

rhubarbdog, 2018-12-17:
The issues for this may be subtle. If i flash my microbit `uflash test_fs.py` I get `OSError: 28` on the first run. but if I flash the microbit `uflash` then (using microfs) `ufs put test_fs.py`. I can run the program once

closed 2018-12-17 by rhubarbdog