bbcmicrobit/micropython #614: neopixel glitches

rhubarbdog, 2018-12-06:
i've got an 8x8 matrix of kitrinik zip LED (neopixel [ws2812] compatible addressable LEDs).

This is wierd because only program neobug3.py shows this issue.
if the code from `draw-screen` is in the main loop and not in a function - no issue
if the code only calls `zip_led.show()` once per `zip_led.clear()` and redraw - no issue.
`zip_led.show()` has to be called twice per cycle once in the main loop and once in a function

I have (tried) to enclose a short video
[neobug3.py.txt](https://github.com/bbcmicrobit/micropython/files/2652925/neobug3.py.txt)

[20181206_112639.mp4.gz](https://github.com/bbcmicrobit/micropython/files/2652939/20181206_112639.mp4.gz)

dpgeorge, 2018-12-07:
What version of micro:bit MicroPython do you use?  Can you attach the exact .hex file that you use that shows this bug?  It may be you are affected by #606

dpgeorge, 2018-12-07:
Thanks for providing the .hex file.  Upon inspection of it, it looks like it is *not* affected by #606.  So this issue requires further investigation.