bbcmicrobit/micropython #616: The REPL documentation is slightly incorrect

rhubarbdog, 2018-12-07:
On this page https://microbit-micropython.readthedocs.io/en/latest/devguide/repl.html
in both the mac section and the linux section it refers to program `screen` to access the REPL.
In the mac section it states type CTRL-A  + CTRL-D to exit. This isn't correct. To terminate screen you type CTRL-A "\" and answer Y to the question.
Typing CTRL-A + CTRL-D detaches screen leaving REPL active collecting output from `print` statements.  To reattatch you issue command `screen -r`, using command `screen /dev/ttyACM0 115200` results in a rather unhelpful output `[screen is terminating]` and no REPL session.

The mac section should be updated and the Linux section should also include this screen termination information.

It would be nice to include something similar for `picocom` but i don't use it so can't help.

microbit-carlos, 2018-12-14:
Thank you @rhubarbdog, I tested the commands as you described and you are right, we should update the docs. Would you like to create a PR with this change? We'll be happy to help you through the process.

_(Just for clarification, I think the markdown in your message doesn't display the commands correctly. To exit the session and close the connection we need `Ctrl+A` followed by `Ctrl+\`.)_

rhubarbdog, 2018-12-14:
Yes i'll happily do the pull request if you give me some guidance.  Just to be clear it's CTRL+A, plain-backslash, answer Yes to the question

microbit-carlos, 2018-12-14:
When I tested in macOS mojave I had to use `Ctrl+\`, not just `\` (and then Yes to the question).

Have you created a GitHub PR before? If not this GitHub docs should be able to help: https://help.github.com/articles/creating-a-pull-request/

We use [Sphinx](http://www.sphinx-doc.org) to generate the docs, which you can install locally if you would like to preview them, which is recommended, but not mandatory. If you are familiar with [restructuredtext](http://www.sphinx-doc.org/en/master/usage/restructuredtext/) you should be able to start writing them right away. The markup is not that complex and you can have a look at the rest of the docs to see how it's been done already.

rhubarbdog, 2018-12-14:
Can you detatch from a screen with CTRL+A then CTRL+D. Leaving screen collecting your debug output and re-attatch later with ` screen -r`

rhubarbdog, 2018-12-19:
Done. PR submitted

closed 2018-12-19 by rhubarbdog