bbcmicrobit/micropython #617: Radio is incmpatible between blocks and micropython

rhubarbdog, 2018-12-07:
This is probably  a bug with Blocks I've logged it there <a href="https://github.com/Microsoft/pxt-microbit/issues/1712">#1712</a>

There is a lack of compatibility between blocks radio and micropython radio.

**Describe the solution you'd like**
When I have a micropython producer and a blocks consumer. I broadcast on channel 7 group 1 "Message" the consumer logs the message's arrival by executing `on RadioReceive` block but the message logged is blank.
When i have a blocks producer and a micropython consumer. I broadcast on group 1 and receive on channel 7 group 1.  The message is preceded by garbage. On the microbit it displays as `?<???????Hello` or `????F????Hello` . `screen` a serial terminal collects the following
```
        FHello
<Hello
```
I'd like to be able to produce Radio applications in both blocks and micropython that can communicate with each other

ukBaz, 2018-12-07:
FYI:
I did a blog post about the differences between MakeCode and MicroPython radio message structure.
https://ukbaz.github.io/howto/ubit_radio.html

DavidWhaleMEF, 2018-12-07:
This is not a bug. It is a difference in implementation. The two languages were mostly developed in isolation, and the radio formats in the very early days were not well documented, so different implementations appeared as a result.

FYI: I wrote a knowledge-base article about this, with code examples of why there are differences and ways to inter-work.

https://support.microbit.org/support/solutions/articles/19000053168-receiving-radio-data-from-the-makecode-editor-within-python

microbit-carlos, 2018-12-14:
As @ukBaz and @DavidWhaleMEF have covered (thanks!) this is due to the different implementation of the radio  module.
The articles linked in those responses should give a pretty good overview on how to get MicroPython and MakeCode programs talking with each other via radio.

I'll close this issue, but feel free to reopen or add additional comments if you have any other questions.

closed 2018-12-14 by microbit-carlos