bbcmicrobit/micropython #619: Can't install compilation packages - ubuntu 16.04

rhubarbdog, 2018-12-12:
Hi,
I want to be able to compile my own firmware, but i am using a ubuntu 16.04 clone. I've been following the discussion <a href="https://github.com/bbcmicrobit/micropython/issues/604">#604</a> but it seem's that the OP has dropped out of the discussion.
I follow the reasoning that `ppa/ppamiller` is no longer required and procede to follow the instructions in README.md.
when i type `sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord libssl-dev` it terminates with the error
```
Errors were encountered while processing:
 /var/cache/apt/archives/binutils-arm-none-eabi_2.26-4ubuntu1+8_amd64.deb
 /var/cache/apt/archives/gcc-arm-none-eabi_15%3a4.9.3+svn231177-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

carlosperate, 2018-12-12:
Could you try to run this and see if it works on its own?

```
sudo add-apt-repository -y ppa:team-gcc-arm-embedded/ppa
sudo apt-get update -qq
sudo apt-get install -y gcc-arm-embedded
```

rhubarbdog, 2018-12-12:
> Could you try to run this and see if it works on its own?
>
> ```
> sudo add-apt-repository -y ppa:team-gcc-arm-embedded/ppa
> sudo apt-get update -qq
> sudo apt-get install -y gcc-arm-embedded
> ```

all worked with no errors to reports

when i try `sudo apt-get install cmake ninja-build gcc-arm-none-eabi srecord libssl-dev` it reports

```
Errors were encountered while processing:
 /var/cache/apt/archives/binutils-arm-none-eabi_2.26-4ubuntu1+8_amd64.deb
 /var/cache/apt/archives/gcc-arm-none-eabi_15%3a4.9.3+svn231177-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

carlosperate, 2018-12-14:
Hi @rhubarbdog,

Doing `apt-get install gcc-arm-embedded` should get you the GCC ARM compiler already, so there is no need to include `gcc-arm-none-eabi` in the next command. Can you try instead:

```
sudo apt-get install cmake ninja-build srecord libssl-dev
```

So, in summary, it would be:

```
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt-get update
sudo apt-get install gcc-arm-embedded
sudo apt-get install cmake ninja-build srecord libssl-dev
```

rhubarbdog, 2018-12-14:
That has appeared to work. Nothing new was added this time so the last attempt must have had an element of success.  What now?
Continue to follow the instructions in the README.md?

rhubarbdog, 2018-12-17:
hi
I tried `sudo -H pip3 install yotta` and got these errors. I notice one if for `pywinusb` i'm using linux is that not microsoft?

```
pyocd 0.14.1 requires enum34, which is not installed.
pyocd 0.14.1 requires hidapi, which is not installed.
pyocd 0.14.1 requires pywinusb>=0.4.0, which is not installed.
pygithub 1.43.3 has requirement requests>=2.14.0, but you'll have requests 2.9.1 which is incompatible.
```

carlosperate, 2018-12-18:
Did the error messages stop the installation? Or was yotta still installed and accessible via command line? I think you can ignore the `pywinusb` error/warning as it should only install on Windows.

If Yotta is not installed, could you copy/paste the entire `pip intall` output?

rhubarbdog, 2018-12-19:
i type `mkdir bbc-microbit`then `cd bbc-microbit` 
`yt target bbc-microbit-classic-gcc-nosd` nothing is output which means `yotta` has sufficently installed otherwise `yt ....` would result in error `yt: command not found`.
I then type `yt up` and get error messages
```
error: No module.json file.
error: The current directory does not contain a valid module.
```
i'm unable to provide output from `pip3` without typing something like `sudo -H pip3 install --no-cache --upgrade yotta`

rhubarbdog, 2018-12-19:
I include missing steps:
1) Download project by clicking `Download or Clone` button above
2) unzip micropython-master
3) change to that directory `cd micropython-master`
4) `yt up`

rhubarbdog, 2018-12-19:
typing `make all` results in following errors
```
/home/phil/bbc-microbit/micropython-master/yotta_modules/mbed-classic/api/FileBase.h:40:31: fatal error: sys/syslimits.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
Makefile:19: recipe for target 'yotta' failed
make: *** [yotta] Error 1
```

rhubarbdog, 2018-12-19:
got it to work.
produced a micropython-master/build/firmware.hex and tried it on a program it worked
There is incorrect info in README.md and some missing steps and information omitted .
Do you want me to update README.md with my/our findings?

1lzx, 2019-06-11:
how to relax

1lzx, 2019-06-11:
/home/phil/bbc-microbit/micropython-master/yotta_modules/mbed-classic/api/FileBase.h:40:31: fatal error: sys/syslimits.h: No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
Makefile:19: recipe for target 'yotta' failed
make: *** [yotta] Error 1

/**/
Could you tell me how to solve this problem?

rhubarbdog, 2019-06-11:
See the README.md in repository https://github.com/rhubarbdog/micropython that has good information on install. Follow the steps exactly