bbcmicrobit/micropython #622: Issuses with set_pull on pin5/pin11 with latest firmware.hex

rhubarbdog, 2018-12-19:
Last night I cloned the repository and compiled my first firmware.hex this morning.
I wanted to add pin5 and pin11 support to some assembly routines i've written
`pinX.set_pull(pinX.PULL_DOWN)` does nothing for button pins (pin5 & pin11)

when i run this
```
from microbit import *

pin5.set_pull(pin5.PULL_DOWN)
pin11.set_pull(pin11.PULL_DOWN)

print("pin5", pin5.read_digital())
print("pin11", pin11.read_digital())
```

i get the following output
```
pin5 1
pin11 1
```

is there internal hardwired pull up resistors on pins 5 an 11?

dpgeorge, 2018-12-19:
> is there internal hardwired pull up resistors on pins 5 an 11?

Yes it is 10k Ohm.  See the bottom of this doc page: https://microbit-micropython.readthedocs.io/en/v1.0.1/pin.html

rhubarbdog, 2018-12-19:
Is that why  `pin5.write_digital(0)` produces an exception `ValueError pin in button mode`. The pin can't be an output only a button or read digital with `pinX.set_pull(pinX.PULL_UP)`

dpgeorge, 2018-12-20:
> The pin can't be an output only a button or read digital

Yes.  See #556 for discussion about this.

closed 2018-12-20 by rhubarbdog