bbcmicrobit/micropython #636: USB HID Keyboard/Mouse Feature Request

scruss, 2019-01-30:
I realize it's complex with the micro:bit's separate USB controller, but for our work with assistive technology, having USB HID would be immensely helpful. Even if BLE worked with MicroPython on the micro:bit, it's not very reliable in many AT situations: disconnections and pairing are just not acceptable to most clients. 

With the micro:bit's sensors, HID would help us to develop inexpensive controllers for more people with disabilities. Right now we're mostly sticking with Adafruit Trinket M0s that really only have three or four button inputs.

(I work for [Makers Making Change](https://www.makersmakingchange.com/): we connect makers to people with disabilities who need assistive technologies. We're a not-for-profit, and our projects are open source)

WayneKeenan, 2019-01-30:
These links might help you?

https://github.com/lancaster-university/microbit-dal/issues/233

https://github.com/kshoji/pxt-bluetooth-keyboard

scruss, 2019-02-04:
Thank you - it's good to know that others are using the micro:bit for accessibility. These seem to be BLE projects, though. I'm specifically asking about USB, as our user tests have found BLE to be too unreliable, especially around power wheelchairs with multiple users in the same room.

closed 2019-02-04 by scruss