bbcmicrobit/micropython #650: spi.read() is non-blocking

rhubarbdog, 2019-03-30:
on the pyboard SPI.recv() blocks and if nothing is received times out with an error OSError ETIMEDOUT.  On the microbit spi.read() just immediately returns a bytearray `b'\x00' * nbytes`
could it block or alternatively return `None`

deshipu, 2019-03-30:
It's not physically possible. The way SPI mater works, it toggles the clock pin and reads the value from the data pin. The slave sets the data pin to high or low on the clock edge, and that's it. There is no way for the master to know whether the pin has not been set by the slave, or whether the slave is simply sending a 0. There is no flow control.

deshipu, 2019-03-30:
In other words, the slave is always sending something. There is no concept of "nothing is received".

closed 2019-03-31 by rhubarbdog