bbcmicrobit/micropython #652: ustruct module not documented

whaleygeek, 2019-03-31:
As per a discussion on slack just now...

This search does not find anything about the ustruct module...

https://microbit-micropython.readthedocs.io/en/latest/search.html?q=ustruct&check_keywords=yes&area=default

[image]

But using the latest MicroPython from https://python.microbit.org/v/1.1

It does exist in the runtime image...

[image]

microbit-carlos, 2019-03-31:
Thanks David, we should port the docs (or link to) from https://docs.micropython.org/en/latest/library/ustruct.html

rhubarbdog, 2019-04-01:
What other modules are there?
ujson ?
uhashlib ?

Would it be possible to create an alphabetical module index as per the micropython docs. Or is that a lot of work for just a few modules which are indexed elsewhere

microbit-carlos, 2019-04-01:
You can see the available modules running `help('modules')`:

```
MicroPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit v1.0.1 with nRF51822
Type "help()" for more information.
>>> help('modules')
__main__          love              os                time
antigravity       machine           radio             ucollections
array             math              random            ustruct
audio             microbit          speech            utime
builtins          micropython       struct
collections       music             sys
gc                neopixel          this
Plus any modules on the filesystem
>>> 
```

The API Reference index on the left-side menu should have an entry for each one of these, excluding aliases (e.g. time/utime), and easter eggs (antigravity, love, and this).

I've created https://github.com/bbcmicrobit/micropython/issues/655 to track this, and PRs are welcomed :)