bbcmicrobit/micropython #655: Missing modules in the docs

microbit-carlos, 2019-04-01:
The following MicroPython modules are not referenced in the docs:

* array
    - [Upstream docs](https://docs.micropython.org/en/latest/library/array.html)
* builtins
    - [Upstream docs](https://docs.micropython.org/en/latest/library/builtins.html)
* ucollections
    - [Upstream docs](https://docs.micropython.org/en/latest/library/ucollections.html)
* gc
    - [Upstream docs](https://docs.micropython.org/en/latest/library/gc.html)
* math
    - [Upstream docs](https://docs.micropython.org/en/latest/library/math.html)
* ustruct
    - [Upstream docs](https://docs.micropython.org/en/latest/library/ustruct.html)
    - Related issue: https://github.com/bbcmicrobit/micropython/issues/652
* sys
    - [Upstream docs](https://docs.micropython.org/en/latest/library/sys.html)

We should either port the upstream docs (note that the module might only be partially enabled in the micro:bit port) or add an entry and link to the upstream docs.
Taking in consideration we have already ported `utime`, `micropython`, and `machine`, we should probably do the same for these.

mofada, 2020-11-02:
Running `help('modules')` in REPL mode shows that there is a utime module, but I cannot find it in the editor, what should I do?

[image]

[image]

mofada, 2020-11-02:
Oh, sorry, I think my problem is a plug-in problem

kopp, 2020-11-02:
I guess, that you are using this package https://pypi.org/project/pseudo-microbit/ which does not provide `utime`. Maybe you can open an issue in this project?

mofada, 2020-11-02:
@kopp Yes, I am using this plugin, but it seems that the most recent update was three years ago and I am not sure if it is still being maintained.

mofada, 2020-11-02:
[intellij-micropython](https://github.com/vlasovskikh/intellij-micropython)  Or this plug-in is also very good, but it also lacks those few files. I filed an issue and I don’t know when it will be updated.