bbcmicrobit/micropython #656: Firmware build fails on MacOS at make echo command

diaena, 2019-04-04:
`make` generates an incorrect `VER_ADDR_FILE`: 

[image]

[image]

(It should be of the form `0x00036df2`)

This is because make uses `sh` to execute commands by default, but MacOS and Linux implement `sh` differently.
On MacOS, `sh` functionality is implemented by bash which on MacOS is compiled with `--enable-strict-posix-default` and as a result of this flag is that the `echo` command does not understand the `-n` flag.

Ways of fixing:

1. use make inbuilt echo command
2. use printf constructs instead of echo (probably nicest and most portable)
3. use the full path to `/bin/echo`

WayneKeenan, 2019-04-04:
I just raised issue #657 and PR #658, which also fixes an `objdump` command issue,

Frederic-soft, 2021-03-18:
There is also an issue with ```size```, which should be ```arm-none-eabi-size``` on line 16 of the Makefile.

dpgeorge, 2021-09-06:
Fixed by dc92faf91be4cdf74130ec9144a5e5ba02d3836d and 4cd8ad10529eeec9b58b9d9ad9f39a2b177a1a35

closed 2021-09-06 by dpgeorge