bbcmicrobit/micropython #661: music.pitch latches when music.stop occours on a different pin

rhubarbdog, 2019-05-15:
I have this program which plays 440Hz on `pin2`. I stop `pin0` with `misic.stop()`.  The short note playing on `pin2` latches.
[latch03.py.txt](https://github.com/bbcmicrobit/micropython/files/3182548/latch03.py.txt)

dpgeorge, 2019-05-17:
The music module can only handle playing music on one pin at a time, it's not supported to mix different pins in calls to play/pitch/stop/etc.

For the test script attached above, what happens is that the PWM is started on pin2 and then `music.stop()` stops the music subsystem and cancels the PWM on pin0.  But pin2 continues to have PWM output on it and will never be turned off (unless it's used again) because the music subsystem is now idle.

I don't think anything really needs to be fixed here.  Rather the docs could be updated to mention this, that you must use the same pin for all subsequent calls to the music module.

rhubarbdog, 2019-05-17:
Why isn't the 12 millisecond time not expiring on `ubit.pin2`?

dpgeorge, 2019-05-17:
> Why isn't the 12 millisecond time not expiring on `ubit.pin2`?

Because when `music.stop()` is called it stops everything related to music, including the timeout for pin2.  So there is never any event which stops PWM on pin2.