bbcmicrobit/micropython #662: This program example iw wrong

rhubarbdog, 2019-05-15:
I'm looking at the page for random https://microbit-micropython.readthedocs.io/en/latest/tutorials/random.html
there is this program example of a dice
```
from microbit import *
import random

display.show(str(random.randint(1, 6)))
```
further down the page random seed is intorduced with this program for a dice
```
from microbit import *
import random

random.seed(1337)
while True:
    if button_a.was_pressed():
        display.show(str(random.randint(1, 6)))
```
The page then goes on to say "Can you work out why this program needs us to press button A instead of reset the device as in the first dice example..?"

when i set the seed to 123
I press `button_a` to release a number and always get the sequence 6,1,1,1,1,1

The program should be
```
from microbit import *
import random

random.seed(123)
while True:
    number = random.randint(1, 6)
    if button_a.was_pressed():
        display.show(str(number))
```

dpgeorge, 2019-05-17:
With a seed of 123 I get the sequence: 6, 1, 1, 1, 1, 1, 1, 1, 1, 2, 3, 4, 5, 1, 3, ...

So I don't think there's anything wrong.

rhubarbdog, 2019-05-17:
Is it not the case that the phenomenon of setting a seed and always getting the same sequence should be shown with this code 
```
for _ in range(2):
random.seed(123)
    for i in range(7):
        print(random.randint(1,6))
```

I think to use seed with a dice example gives poor indication of what is trying to be demonstrated

dpgeorge, 2019-05-17:
I think the existing tutorial is fine.  It shows how to make the device have the same set of random numbers after it is reset.