bbcmicrobit/micropython #663: How to log local on microbit?

davorradic, 2019-05-28:
I have next code:
`import time
from microbit import *
def writeToLocalStrage():
    with open('log.txt') as old_file:
        text_from_file_before = old_file.read()
    
    with open('log.txt', 'w') as my_file:
        display.show(text_from_file_before)
        content = str(temperature())
        content = text_from_file_before + content
        content = content + ';'
        my_file.write(content)
    
    
while True:
    writeToLocalStrage()
    time.sleep(2)`

if I want to get it with:
ufs get log.txt
I get only 3x
28;28;28;
but never more then this, is the memory full or my script is not good?

dpgeorge, 2019-06-03:
The above code works fine for me, it wrote up to at least 10 entries (and then I stopped it).

I suggest to check what other files you have on the microbit's filesystem.

yoonghm, 2019-06-18:
Try this:

```Python
from microbit import *
import time

with open('log.txt', 'a') as f:
    display.show(f.read())

while True:
    with open('log.txt', 'a') as f:
        my_file.write(str(temperature(),';')
    time.sleep(2)
```